arduino/Arduino #6753: Please add library from MOVI Speech Recognition Shield

fractor, 2017-09-23:
Our MOVI library has been open source from the beginning and it's available on github.
https://github.com/audeme/MOVIArduinoAPI/releases/tag/v1.1.13

Several users requested download capability in the IDE. So let's make it happen :-) Let me know if I need to fix anything.

per1234, 2017-09-23:
The `version` value in library.properties must match the version of your release. The `version` value in your library.properties is [1.13](https://github.com/audeme/MOVIArduinoAPI/blob/v1.1.13/library.properties#L2) but your release version is[ 1.1.13](https://github.com/audeme/MOVIArduinoAPI/releases/tag/v1.1.13).

Please either re-do your release or create a new one with matching versions. Once you have done so, comment here to let us know we can proceed with adding your library to the index.

audeme, 2017-09-24:
Thanks for noticing. I re-did the release with updated library.properties file. 
https://github.com/audeme/MOVIArduinoAPI/releases/tag/v1.1.13

cmaglie, 2017-09-29:
@fractor 
we do not allow parenthesis in the library name: `MOVI(tm) Voice Dialog Shield` to fix this:

- remove the release tag
- update the name in the library.properties
- commit & push
- re-create the release tag

thanks for your patience! :-)

audeme, 2017-09-29:
Done. Updated URL:
https://github.com/audeme/MOVIArduinoAPI/releases/tag/v1.1.13

I am curious though because locally the parentheses were no problem.

cmaglie, 2017-10-02:
Done!

cmaglie, 2017-10-02:
> I am curious though because locally the parentheses were no problem.

There is no problem locally with the latest releases of the IDE because we have recently relaxed the checks on the names. Anyway users that had a previous release installed may experience some problems: for this reason we decided to restrict the characters allowed in the library manager to maximize backward compatibility.

closed 2017-10-02 by cmaglie