arduino/Arduino #6754: IDE not being displayed on monitor

RadarRon, 2017-09-23:
I’ve got a problem with the IDE. I **used** to have 2 monitors on my PC and now I don’t. The last time I used the IDE it was showing on the **now** non-exitance display. So now when I open the IDE windows __(ver 10_)_ shows that it’s open and running but I can’t see it, so somehow the IDE is trying to display its self on that missing monitor. 

> So my question is...

**How can I move the IDE to my screen if I can't click and drag it?**

feikname, 2017-09-23:
@RadarRon Try to press Win + LeftArrow (with the Arduino window being the currently active one). The Windows might come back, snapping to the left side of your main monitor.

RadarRon, 2017-09-23:
Thanks, Feikname, but that was a when I did as instructed nothing happened on my screen. I tried **alt+tab** and it brought up all the active windows but when I selected the IDE still nothing on the screen.  Then tried **Win+tab** then right clicked on the IDE. There is an option to move it between desktops. Desktop 2 is grayed-out. tryed moving it and still nothing.

I finally gave up and deinstalled it and reinstalled it. works now

**_WARNING_** to anyone trying this, copy all your sketches and any libraries you have made to someplace safe the deinstall will destroy all your hard work.

facchinm, 2017-09-25:
Hi @RadarRon ,
the problem you are experiencing should be partially solved by https://github.com/arduino/Arduino/commit/8292989e2dab39c62d03ddd9c25b7f4570813b7d, included in 1.8.3. Next time, instead then uninstalling and reinstalling (which never messes with your sketchbook, so you can do it safely) you can simply modify `preferences.txt`, removing all `last.*` entries, to restore the default properties.

closed 2017-09-25 by facchinm