arduino/Arduino #6758: wierd error

zosamohak, 2017-09-24:
Arduino : 1.8.4 (Windows Store 1.8.8.0) (Windows 10), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

c:\program files\windowsapps\arduinollc.arduinoide_1.8.8.0_x86__mdqgnx93n4wtt\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: unable to rename 'core\core.a'; reason: Permission denied

exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

per1234, 2017-09-24:
See:
https://github.com/arduino/Arduino/issues/5616
https://github.com/arduino/Arduino/issues/4683

Please try **TEMPORARILY** disabling your antivirus software for a single compilation to see if that solves the issue.

per1234, 2017-10-25:
@zosamohak did the issue still occur when you disabled the antivirus?

closed 2018-02-28 by per1234

Mayoogh, 2021-04-18:
Temporary fix 
- Create a new blank  sketch 
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
- Verify/Compile 
- Close new sketch and try compiling project sketch