arduino/Arduino #6759: consider adding my library shift7seg. driving 7 segment displays from 74hc595 shift registers

pseudoVella, 2017-09-25:
url=    https://github.com/pseudoVella/shift7seg.git

for now it's just for a 4 digit display but i'm  considering  making it compatible with up to 64 digits. working on a proper schematic to post, but you should be able to figure out the hookup from comments in shift7seg.h file for now.

per1234, 2017-09-25:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

pseudoVella, 2017-09-25:
Oops sorry about that... Okay added a release, should be good to go now .

cmaglie, 2017-09-29:
Done!

closed 2017-09-29 by cmaglie