arduino/Arduino #6760: speed up for printFloat in print.cpp (AVR)

VASilaev, 2017-09-25:
size_t Print::printFloat(double number, uint8_t digits) 
{ 
  
  
  // занизим для экономии памяти
  uint8_t n = 0; 
   
  //Зачем на все эти частности если не можем вывести то просто пишем err
  // по хорошему текст нужно вынести на флеш память тогда еще +4 байта к оперативке
  
  if (((uint16_t)(*((uint32_t *) &number) >> 16) & 0x7f80) > 0x4F00) {
    n = print("err");
  } else {
    
  
    
    // обрабатываем отрицательное значение
    if ((uint8_t)(*((uint32_t *) &number) >> 24) & 0x80) {
       number = -number;
       n = n + print('-');
    };  
      
    // корректировка округления что бы print(1.999, 2) вывелся как "2.00"

    uint32_t rounding = 1;
    
    uint8_t i = digits;
    while (i) {
      rounding = rounding * 10;
      i --;
    };
    number = number + 0.5 / rounding;  
      
    // выводим целую часть
    n  = n + print((unsigned long)(number), DEC);
    

    //Выводим дробную часть если необходимо
    if (digits > 0) {
    
      n = n + print('.'); 
      
      register uint32_t remainder;     
  
        // остаток. магическое число 0xA0000000 смещает выводимое число на позицию с 28 по 31 бит

      number = number - (double)((unsigned long)(number));

      remainder = (unsigned long) (number * 2684354560.0);

      
      //извлекаем по одной цифре
      do  {
        uint8_t toPrint;
        
         // извлекаем число для записии и обнуляем его в остатке
     
        uint8_t t;
        asm volatile (
         "ldi %1, 0x0f" "\n\t"       
         "mov %0, %D2" "\n\t"
         "swap %0" "\n\t"
         "and %0, %1" "\n\t"
         "and %D2, %1" "\n\t"
         : "=r" (toPrint), "=d" (t), "+l" (remainder)
         : 
        );       
        
        n = n + write(toPrint + '0');
        
        if (!--digits) break;
        
        // следующий символ встает на свое место
        remainder *= (uint16_t) 10;
      } while (1);
     
    };

  };

  
  return (uint16_t) n;
};

matthijskooijman, 2017-09-25:
Could you explain what you're changing exactly and why it works, in English? It might be better to open up a PR or provide a diff, since that makes easier to see what changes and what stays the same.

darrylp1per, 2018-09-03:
instead of looping and calculating the rounding value each time you print,  and using a slow division

the following is much faster at the expense of some ram.

`    const double rounding[] = { 0.5, 0.05, 0.005, 0.0005, 0.00005, 0.000005, 0.0000005 };
    number = number + rounding[digits];
`

replacing

`    // корректировка округления что бы print(1.999, 2) вывелся как "2.00"

    uint32_t rounding = 1;
    
    uint8_t i = digits;
    while (i) {
      rounding = rounding * 10;
      i --;
    };
    number = number + 0.5 / rounding;  
`

sandeepmistry, 2019-09-16:
I'm closing this as won't fix, moving forward the Print class will be shared in: https://github.com/arduino/ArduinoCore-API

We should not have platform specific assembly inside it.

closed 2019-09-16 by sandeepmistry