arduino/Arduino #6762: Varable definition preset error

ameliaamelia, 2017-09-26:
with:
uint8_t Var1 = 0; // (zero)

Var1 contains  1

with other vales all OK.

Arduino DUE IDE 1.8.4

per1234, 2017-09-26:
Please provide a complete, minimal sketch that demonstrates the issue,

ameliaamelia, 2017-09-26:
With a simple sketch all is OK, my application has more than 4000 rows of code; all worked fine but when I added some varables, the error rised; I solved putting all initalizations into the Start()

iPhone di Silvio

per1234, 2017-09-26:
Without a minimal demonstration sketch it will be impossible for us to investigate this so I'm closing as invalid.

If you do find a way to produce one in the future that shows this to be a bug with the Arduino IDE then please update this issue and I'll re-open.

closed 2017-09-26 by per1234