arduino/Arduino #6764: Add Request Library

Erhan-MADE, 2017-09-26:
lightweight library for stepper motor, with actions that specify sequential movements and options to trigger callbacks at the [end.](https://github.com/Erhan-MADE/StepperControl)

per1234, 2017-09-26:
There are some things you need to do to make your repository eligible for inclusion in the Library Manager index:
1. Create a library.properties file following the Arduino Library Specification: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
1. Create a release or tag that matches the `version` value set in library.properties.

After you have done this, please comment here so we know we can proceed.

More information: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Erhan-MADE, 2017-09-27:
Added files and created a release,

thanks

per1234, 2017-09-27:
Actually you created a release and then added the library.properties file, which means your release does not contain a library.properties file:
https://github.com/Erhan-MADE/StepperControl/tree/2.0

Please either re-do your release or create a new one and then comment here so we know we can proceed with adding your library to the Library Manager index.

Erhan-MADE, 2017-09-27:
You are right, still not used to using git properly.
Made a new release, bumped the version to 2.0b

sorry for the inconvenience :(

per1234, 2017-09-27:
The `version` value must be semver compliant, which `2.0b` is not. From https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format:
>- version - version of the library. Version should be semver compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

More information: http://semver.org/

Once you have resolved this issue please comment here so we know we can proceed.

Erhan-MADE, 2017-09-27:
this is getting embarrassing -.-
version 2.1 it is

Erhan-MADE, 2017-09-27:
merged your fork, and made a new release

cmaglie, 2017-09-29:
Done!

closed 2017-09-29 by cmaglie