arduino/Arduino #6766: Mac OS 10.13, High Sierra

rongeurcloclo, 2017-09-27:
The application Arduino do not work well this this new OS, i lost all the application menus.

facchinm, 2017-09-27:
Hi @rongeurcloclo , we are keeping track of the issue [here](https://github.com/arduino/Arduino/issues/6548) . We just merged a commit, so you could try downloading the [nightly build](https://www.arduino.cc/en/Main/Software#hourly) as soon as the counter updates and report if it works on the other thread. Thanks!

closed 2017-09-27 by facchinm

rongeurcloclo, 2017-09-27:
I dowload the nightly build but still not working, i can compile but not see the menus

facchinm, 2017-09-27:
The nightly is now live, please retest with the current zip thanks

rongeurcloclo, 2017-09-27:
OK, it seems to work, but the menus are attach to each window, like on PC windows

rongeurcloclo, 2017-09-27:
I can not quit with the menu, but if i close the window, the application quit, but it's working, i tried with Arduino Uno

PaulStoffregen, 2017-09-27:
Tried the latest hourly build just now.  On my English language Macbook Air, the menus still appear normally (not on each Window).  An inconsistent user experience might not be so good?

rongeurcloclo, 2017-09-27:
I'm using arduino on french system and french langage with a macBook pro 2015 and the menu still on each window

facchinm, 2017-09-28:
Thanks for spotting the quit() bug, I just pushed 59a57a9312bd02d7dbc92dc7846f4216f2793c70 to fix it. Thanks again!

rongeurcloclo, 2017-09-28:
I have downloaded the new version, the menus are always in each of the windows, but the program leaves with the menu, and it seems to work well
[image]