arduino/Arduino #6772: New library: Simple log macroses

ploys, 2017-09-29:
Hi,
I have publish a tiny macros-set for logging which use in own projects, could you please add it to Library Manager.

Repository: https://github.com/ploys/arduino-logger/releases/tag/1.0.0

Thanks

cmaglie, 2017-10-02:
There is already a library named `Logger`, you must choose another name for your library.
To fix this:

- delete the current release tag
- edit the `library.properties` file and change the `name` field
- I suggest to change also the name of the include to reflect the new name
- commit & push
- recreate a new release tag

thanks!

ploys, 2017-10-09:
Hi, 
Is it ok if I rename it to Log?

ploys, 2017-10-09:
Lib renamed, new release tag:
https://github.com/ploys/arduino-logger/releases/tag/v1.0.0

ploys, 2017-10-10:
Hi,
I fix fatal error which invoke reboot of esp8266 board, updated release is here:
https://github.com/ploys/arduino-logger/releases/tag/v1.0.1

matthijskooijman, 2017-10-10:
New versions should be automatically picked up by the IDE, but only if you increase the version number in the library.properties file, which you did not do for this tag. The version number in the tag is not used, though you must create a tag for the release to be picked up.

ploys, 2017-10-10:
Thanks for advice, is it all ok now?
https://github.com/ploys/arduino-logger/releases/tag/1.0.1

cmaglie, 2017-10-12:
Done!

closed 2017-10-12 by cmaglie