arduino/Arduino #6773: Arduino IDE always still startup "Starting..." on mac OS 10.13

0x1abin, 2017-09-30:
[image]

`➜  ~ /Applications/Arduino.app/Contents/MacOS/Arduino 
[JmDNS(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).Timer] WARN javax.jmdns.impl.tasks.resolver.DNSResolverTask - ServiceResolver(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).run() exception 
java.io.IOException: No route to host (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1595)
	at javax.jmdns.impl.tasks.resolver.DNSResolverTask.run(DNSResolverTask.java:74)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - RECOVERING
[JmDNS(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).State.Timer] WARN javax.jmdns.impl.tasks.state.DNSStateTask - Canceler(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).run() exception 
java.io.IOException: No route to host (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1595)
	at javax.jmdns.impl.tasks.state.DNSStateTask.run(DNSStateTask.java:131)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Invalid library found in /Users/Zibin/Documents/Arduino/libraries/EasyWebSocket: /Users/Zibin/Documents/Arduino/libraries/EasyWebSocket
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.DNSStatefulObject$DefaultImplementation - Wait for canceled timed out: DNS: fe80-0-0-0-aede-48ff-fe00-1122-en5.local. [fe80:0:0:0:aede:48ff:fe00:1122%en5/fe80:0:0:0:aede:48ff:fe00:1122%en5] state: canceling 1 task: null
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - fe80-0-0-0-aede-48ff-fe00-1122-en5.local.recover() Could not recover we are Down!
2017-09-30 14:08:44.492 Arduino[2952:88867] IMKInputSession [0x7fd69dce65a0 presentFunctionRowItemTextInputViewWithEndpoint:completionHandler:] : [self textInputContext]=0x0 *NO* NSRemoteViewController to client, NSError=Error Domain=NSCocoaErrorDomain Code=4099 "The connection from pid 0 was invalidated from this process." UserInfo={NSDebugDescription=The connection from pid 0 was invalidated from this process.}, com.apple.inputmethod.EmojiFunctionRowItem
`

0x1abin, 2017-10-01:
0 0 help..

emdete, 2021-09-20:
i had this error today. it's still there. it showed up after a successful compile. why does the IDE do dns during compile anyway? i expect it not to do *any* network traffic until i want to download something.

closed 2021-10-14 by 0x1abin