arduino/Arduino #6774: Eagle files out of date for MEGA 2560

tburba, 2017-09-30:
Recently I bought the board labeled "Mega 2560 V3E" on bottom copper.

The official files, https://www.arduino.cc/en/uploads/Main/arduino-mega2560_R3-ref-design.zip, contain definitely different silk screen layers. Top copper reveals "MEGAXL-Rev3b" (absent-mindedly hidden under the power socket). Location of the button "RESET" is totally different (inaccessible when using shields). Looks like my V3E is really improved! Remaining copper traces are fine in a few places I looked at, however it's not possible to be sure. This is distracting, lowers credibility and wastes time, especially for newbies.

By the way, the official photos on https://store.arduino.cc/arduino-mega-2560-rev3 have the button as per Rev3b however the silk screen is already newer. Likely these are "3c".

Surely you keep somewhere every version of the design that went into production, don't you? Then why not _publish every one of them_? It's open hardware after all, and 200 KB is a ludicrously small amount.

kengdahl, 2017-12-27:
Just for clarification: You want links to all the old board designs. Yes?

kengdahl, 2017-12-27:
So basically: The current schematics does not match your newly bought Mega?

And: A archive for old schematics and silks is wanted.

tburba, 2017-12-27:
Correct. Ideally this archive would be updated shortly after every new hardware revision, as otherwise future users will face the same problem.

pespinozav, 2018-08-06:
Are there any updates on this? Is there a place to get the latest Mega design? Is the V3E the latest one?
Thanks.