arduino/Arduino #6775: Cannot rotate through open windows using Command + ~ using OSX High Sierra and 1.8.4

oldmanegan, 2017-09-30:
Tried latest hourly build (9/28), no luck. 
Previously, in 1.8.3 and prior, and with all versions of OSX including High Sierra, one was able to rotate through all open windows using Command + ~ (including using the shift key to rotate in the 'opposite' order). As of 1.8.4 it only rotates through last two opened windows. 
This is fine when one open window is the Monitor, thus ensuring the monitor window and its associated sketch are only rotated through, avoiding possible erroneous changes to other sketches. However, when monitor is not open, it only rotates through last two opened files. Is this a bug or an "improvement?" I suggest keeping the new functionality/logic when monitor open, but going back to previous functionality when monitor not open.

facchinm, 2017-10-02:
Hi @oldmanegan , it looks like a bug but I'm unable to reproduce it (neither on high sierra).
Which keyboard layout are you using? On English OSX, the keyboard shortcut to move between tabs is `alt + cmd + left or right arrow` and it still works correctly on all IDE versions I tested.

per1234, 2017-12-04:
@oldmanegan we're still waiting to hear back from you regarding which keyboard layout you're using.

lmihalkovic, 2018-04-29:
Had the same issue yesterday: could not cycle through all the __open editor windows__ .. it is what prompted me to implement a Window menu to be able access all the open editors

per1234, 2018-04-29:
I just noticed that @oldmanegan was talking about rotating through open **windows** but @facchinm was talking about moving between **tabs**.

lmihalkovic, 2018-04-29:
yup ... dunno when it appeared as a osx behavior.. just saw it yesterday

lmihalkovic, 2018-04-30:
@oldmanegan I think this was a happy coincidence .. 

I just checked other osx apps: 

The is what Pages & Keynotes offer
[image]

This is what Chrome has implemented:
[image]

kosso, 2018-12-09:
Has this still not happened?

aspinos, 2020-05-04:
I used to have this issue as well (I am on macOS 10.14 Mojave, was running version 1.8.9 of the IDE), but updating to IDE version 1.8.12 fixed the issue. If no one else can replicate this, it can probably be marked as resolved.