arduino/Arduino #6776: Command line upload port option does not support symlinks

VictorLamoine, 2017-09-30:
- Arduino 1.8.4
- Linux (ARM ,Raspberry Pi 3)

Arduino is plugged in on the RPi:
```bash
$ ls -l /dev/serial/by-id/usb-Arduino*
lrwxrwxrwx 1 root root 13 sept. 30 23:51 /dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00 -> ../../ttyACM1
```

Compiling/upload with port = `/dev/ttyACM1` works:
```bash
arduino sketch.ino --port /dev/ttyACM1 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Archiving built core (caching) in: /tmp/arduino_cache_94254/core/core_arduino_avr_micro_80d002210ad4f4c4b138a75ac18d4048.a
Sketch uses 6202 bytes (21%) of program storage space. Maximum is 28672 bytes.
Global variables use 396 bytes (15%) of dynamic memory, leaving 2164 bytes for local variables. Maximum is 2560 bytes.
Uploading...
$
```

But with port = `/dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00` it fails:
```bash
$ arduino sketch.ino --port /dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Archiving built core (caching) in: /tmp/arduino_cache_377862/core/core_arduino_avr_micro_80d002210ad4f4c4b138a75ac18d4048.a
Sketch uses 6202 bytes (21%) of program storage space. Maximum is 28672 bytes.
Global variables use 396 bytes (15%) of dynamic memory, leaving 2164 bytes for local variables. Maximum is 2560 bytes.
Uploading...
An error occurred while uploading the sketch
$
```

Seems to me that the problem is the port being a symbolic link.

matthijskooijman, 2017-10-01:
Or is the problem that the symlink's name is not a known serial port name? What happens if you name the symlink `/dev/ttyACM3` or something like that?

Also, it's a bit problematic that no meaningful error message is shown, that needs some looking at by itself too.

facchinm, 2017-10-02:
Hi @VictorLamoine,
I believe that the problem is originated by `avrdude` ; launching the command with `--verbose` should give more meaningful results that we can use to debug the issue. Thanks!

VictorLamoine, 2017-10-02:
```bash
$ arduino sketch.ino --port /dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00 --verbose
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/opt/arduino-1.8.4/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.4/hardware -hardware /home/victor/.arduino15/packages -tools /opt/arduino-1.8.4/tools-builder -tools /opt/arduino-1.8.4/hardware/tools/avr -tools /home/victor/.arduino15/packages -built-in-libraries /opt/arduino-1.8.4/libraries -libraries /home/victor/Arduino/libraries -fqbn=arduino:avr:micro -ide-version=10804 -build-path /tmp/arduino_build_31488 -warnings=none -build-cache /tmp/arduino_cache_153838 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.4/hardware/tools/avr -verbose /home/victor/sketch/sketch.ino
/opt/arduino-1.8.4/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.4/hardware -hardware /home/victor/.arduino15/packages -tools /opt/arduino-1.8.4/tools-builder -tools /opt/arduino-1.8.4/hardware/tools/avr -tools /home/victor/.arduino15/packages -built-in-libraries /opt/arduino-1.8.4/libraries -libraries /home/victor/Arduino/libraries -fqbn=arduino:avr:micro -ide-version=10804 -build-path /tmp/arduino_build_31488 -warnings=none -build-cache /tmp/arduino_cache_153838 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.4/hardware/tools/avr -verbose /home/victor/sketch/sketch.ino
Using board 'micro' from platform in folder: /opt/arduino-1.8.4/hardware/arduino/avr
Using core 'arduino' from platform in folder: /opt/arduino-1.8.4/hardware/arduino/avr
Detecting libraries used...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_31488/sketch/sketch.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_31488/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_31488/preproc/ctags_target_for_gcc_minus_e.cpp"
"/opt/arduino-1.8.4/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_31488/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_31488/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_31488/sketch/sketch.ino.cpp.o"
Compiling libraries...
Compiling core...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/arduino_build_31488/core/wiring_pulse.S.o"
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/arduino_build_31488/core/WInterrupts.c.o"
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/ha [... cut ...]
PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}
PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}
PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}
PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}
PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}
An error occurred while uploading the sketch
$
```

VictorLamoine, 2017-10-02:
The problem is not the symbolic link:

```bash
$ sudo ln -s /dev/ttyACM1 /dev/ttyACM9
$ ls -l /dev/ttyACM9 
lrwxrwxrwx 1 root root 12 Oct  2 20:01 /dev/ttyACM9 -> /dev/ttyACM1
```

```bash
$ arduino sketch.ino --port /dev/ttyACM9 --verbose
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/opt/arduino-1.8.4/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.4/hardware -hardware /home/victor/.arduino15/packages -tools /opt/arduino-1.8.4/tools-builder -tools /opt/arduino-1.8.4/hardware/tools/avr -tools /home/victor/.arduino15/packages -built-in-libraries /opt/arduino-1.8.4/libraries -libraries /home/victor/Arduino/libraries -fqbn=arduino:avr:micro -ide-version=10804 -build-path /tmp/arduino_build_923475 -warnings=none -build-cache /tmp/arduino_cache_14886 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.4/hardware/tools/avr -verbose /home/victor/sketch/sketch.ino
/opt/arduino-1.8.4/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.4/hardware -hardware /home/victor/.arduino15/packages -tools /opt/arduino-1.8.4/tools-builder -tools /opt/arduino-1.8.4/hardware/tools/avr -tools /home/victor/.arduino15/packages -built-in-libraries /opt/arduino-1.8.4/libraries -libraries /home/victor/Arduino/libraries -fqbn=arduino:avr:micro -ide-version=10804 -build-path /tmp/arduino_build_923475 -warnings=none -build-cache /tmp/arduino_cache_14886 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.4/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.4/hardware/tools/avr -verbose /home/victor/sketch/sketch.ino
Using board 'micro' from platform in folder: /opt/arduino-1.8.4/hardware/arduino/avr
Using core 'arduino' from platform in folder: /opt/arduino-1.8.4/hardware/arduino/avr
Detecting libraries used...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_923475/sketch/sketch.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_923475/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_923475/preproc/ctags_target_for_gcc_minus_e.cpp"
"/opt/arduino-1.8.4/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_923475/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/tmp/arduino_build_923475/sketch/sketch.ino.cpp" -o "/tmp/arduino_build_923475/sketch/sketch.ino.cpp.o"
Compiling libraries...
Compiling core...
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/arduino_build_923475/core/wiring_pulse.S.o"
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.4/hardware/arduino/avr/variants/micro" "/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/arduino_build_923475/core/WInterrupts.c.o"
"/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AV [... cut ...]
Writing | ################################################## | 100% 0.37s

avrdude: 3618 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_923475/sketch.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_923475/sketch.ino.hex:
avrdude: input file /tmp/arduino_build_923475/sketch.ino.hex contains 3618 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.08s

avrdude: verifying ...
avrdude: 3618 bytes of flash verified

avrdude done.  Thank you.

```

VictorLamoine, 2017-10-02:
Maybe `avrdude` makes some assumptions on a COM port name?

VictorLamoine, 2017-10-02:
In the mean time, this works:
```bash
arduino sketch.ino --port $(readlink -f /dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00)
```

matthijskooijman, 2017-10-03:
> Maybe avrdude makes some assumptions on a COM port name?

Not avrdude, but the Arduino IDE. The lines:

    PORTS {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } / {/dev/ttyACM0, /dev/ttyACM1, /dev/ttyAMA0, } => {}

Are generated by the Arduino IDE, which waits for the comport to become available before starting avrdude. The reason is that you're using an Arduino micro, which needs some special twiddling to make it reset into the bootloader. Because USB is builtin on the main chip, this also resets the USB connection, making the comport disappear and re-appear. The IDE waits for this before starting avrdude.

However, AFAIU this does not just check the specified com port, but instead monitors the list of available comports (as generated by the JSSC serial library, based on some regexes I suspect) and sees if any new ones appear (because after the reset, the micro might get a different port assigned than before, e.g. when the port is in use).

However, looking at the "broken" output, the line "Forcing reset using 1200bps open/close on port /dev/ttyACM9" is missing, which makes me suspect that it waits with the reset until the specified port is present in the JSSC port list. However, at least on Linux and OSX, it would probably be better to just check if the specified file exists (and if it's a link, if its target exists) instead of checking the JSSC port list. Alternatively, the IDE could resolve any symlinks and then continue processing with the actual port name.

VictorLamoine, 2017-10-03:
This goes beyond my understanding of the Arduino programs :)

facchinm, 2017-10-05:
@matthijskooijman the touch port routine just waits for **any** port to reappear and uses it as upload port.
I believe that the code that fails is in fact the [touch](https://github.com/arduino/Arduino/blob/101677bba23b6da6d55a46b89d95be0167d47d83/arduino-core/src/processing/app/Serial.java#L98). I need to test on sysfs symlinks to understand if any action on the port might fail.

sintezcs, 2024-02-29:
@VictorLamoine 
> # Alternative
> ```shell
> arduino sketch.ino --port $(readlink -f /dev/serial/by-id/usb-Arduino_LLC_Arduino_Micro-if00)
> ```

Thank you a lot for this! It saved me a ton of time! 🙏