arduino/Arduino #6780: Arduino Due Flash Page Locked Issue...

pdborhad, 2017-10-01:
Hi,

Please help me in resolving the error while programming Arduino Due board. The error is "Flash page is locked". I have tried pressing erase button while programming; however, when you press the erase button, the program loads, but does not work, as I am not able to see print statements on serial window.
Please help me here as more than 5 - 6 board have gave me this issues.

facchinm, 2017-10-02:
Hi @pdborhad,
there is probably something wrong with your code which is hanging the boards. Try uploading a simple sketch (like Blink) and check if the same problem appears.
If the problem still appears, open an issue on the proper repository (https://github.com/arduino/ArduinoCore-sam) with the verbose output of compilation and upload, specifying everything you are using (operating system, IDE version, sam core version and so on).
Thanks!

closed 2017-10-02 by facchinm

JohnJebb, 2019-03-05:
I now have two Due's that show Flash Page Locked errors. This issue appears for no reason on code that has been working in the past. With this problem present, I cannot load any new software to the Due.
I have developed code for Uno's for a while and this has not happened with them.
What can I do about this issue?
I there a fix for this ?