arduino/Arduino #6782: Scrolling Issue

zzcrow, 2017-10-01:
All Linux versions when having lots of examples or libraries to scroll to select, if they go off the screen there is no scrolling past to select them successfully!!! I have a lot of libraries and cannot use Linux versions due to this issue. Someone fix this. Windows and Mac versions work just fine. It may be a native Linux problem though. Thanks

per1234, 2017-10-01:
Which version of the Arduino IDE are you using with Linux?

zzcrow, 2017-10-01:
Running Linux MintArduino IDE version: 2:1.0.5+dfsg2-4

per1234, 2017-10-01:
That's your problem. When you install the Arduino IDE via `apt get` you get a very outdated and possibly non-standard version of the Arduino IDE. You should always download the IDE from https://www.arduino.cc/en/Main/Software no matter which operating system you're using.

closed 2017-10-01 by per1234