arduino/Arduino #6785: Not return actual value from calculating 64 bit int

Mizanurrahman2010, 2017-10-02:
The return value should be **8**
But it return 7. Why???

=========================================
void setup() {
uint64_t num1 = PowerE(77,7);
uint64_t num2 = num1 - (num1/15) * 15);
uint32_t num3 = num2-num1;

Serial.begin(115200);
Serial.print(num3);

}

void loop() {
  // put your main code here, to run repeatedly:

}
long PowerE(long a, long b) {
  int64_t temp = 1;
  for (long i = 0; i < b; ++i) {
    temp = temp * a;
  }
  return temp;
}

facchinm, 2017-10-02:
This issue tracker is only to be used to report bugs or feature requests. This is **NOT** an appropriate place to share your sketch or ask for help. You are welcome to post to the Arduino Forum.

Before doing so, please read http://forum.arduino.cc/index.php?topic=148850
Then create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2017-10-02 by facchinm

matthijskooijman, 2017-10-02:
As @facchinm says indeed. I was having a look at your problem anyway, but the code you pasted won't compile (unmatched parenthesis in the `num2` calculation) and AFAICS it should print `8 - num1`, not 8. Be sure to fix your question when asking on the forum.

Mizanurrahman2010, 2017-10-02:
Solved;
void setup() {
uint64_t num1 = PowerE(77,7);
uint32_t num2 = num1 - ((num1/15) * 15);

Serial.begin(115200);
Serial.print(num2);

}

void loop() {
  // put your main code here, to run repeatedly:

}
uint64_t PowerE(long a, long b) {
  uint64_t temp = 1;
  for (long i = 0; i < b; ++i) {
    temp = temp * a;
  }
  return temp;
}