arduino/Arduino #6787: I want to add my library to Library Manager

maykon, 2017-10-03:
Hi, I want to add my library to Library Manager.

Please check my project: https://github.com/maykon/ButtonDebounce

Thanks.
Maykon L. Capellari

per1234, 2017-10-03:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

maykon, 2017-10-03:
Create release with tag 1.0.1.

Thanks.

maykon, 2017-10-04:
Hi, I need change anymore or it´s OK?

Thanks.

per1234, 2017-10-04:
It's fine now. That's why I removed the "Waiting for feedback" label. cmaglie will add your library to the index after some days when he goes through and does all the outstanding requests. He will close this issue at that time, and you'll get a notification about this. Shortly after that your library will be available for installation via Library Manager.

cmaglie, 2017-10-05:
Done

closed 2017-10-05 by cmaglie