arduino/Arduino #6788: JSSC with Windows 10 (64 bit) and Java 9

wholder, 2017-10-03:
I'm posting this as a possible "heads up" for a potential future problem for the Arduino IDE.  I use JSSC for serial I/O with Java and some time back Arduino switched to JSSC (from RXTX) for its serial I/O, so I think a problem I just started having with JSSC on Windows 10 (64 bit) with the new Java 9 release may potentially be an issue for the Arduino IDE in the future.  For details, see the JSSC wiki entry I posted here:

  https://github.com/scream3r/java-simple-serial-connector/issues/116

I've included some simple Java test code in the wiki entry that demonstrates the problem but, basically, a call to `openPort()` to open a serial connection just completely crashes the JVM.  I'm not an expert at reading Windows stack traces, but I think the issue my result from this call:

`  DCB *dcb = new DCB();`

in `Java_jssc_SerialNativeInterface_openPort()`.

Wayne

facchinm, 2017-10-05:
Hi @wholder ,
thanks for the heads up! We are not planning any switch to Java9 in the upcoming months but of course it could still be a problem if you want to compile the IDE by yourself.

wholder, 2017-10-05:
I've managed to create what I think is a reasonably good fix for the problem.  It boils down to the way memory was being allocated on the stack in the C++ functions `Java_jssc_SerialNativeInterface_openPort()` and `Java_jssc_SerialNativeInterface_readBytes()`.  I was able to build a new version of the `jSSC-2.8_x86_64.dll` file using Visual Studio 2017 and splice it into the `jssc.jar` library file and it seems to work with Java 9 and Windows 10 (at least in the application in which I'm using JSSC.)  However, I'm not really sure how to rebuild all the other native code modules.  I'm also not an expert in writing C++ driver code or Windows, so it would helpful if someone else could vet my code. which I've attached below.

Caveat: I found it rather difficult to test my changes to the DLL file as it seem like it would get cached in some cases and I could myself testing what I thought was new code only to find windows using the prior version.  To detect this. I would temporarily change the value of the `jSSC_NATIVE_LIB_VERSION`  `#define` in the `jssc_SerialNativeInterface.h` file so that I could verify if the new DLL had loaded.

Wayne

[jssc.zip](https://github.com/arduino/Arduino/files/1360960/jssc.zip)

astout, 2017-10-09:
I apologize for posting here as it's not entirely related to Arduino, but I have a bootloader that is written in Java that is using jssc and crashes on Windows 10. @wholder, I tried your method of rebuilding with the modified `jssc.c++` in Visual Studio (VS). The build was successful, but I am concerned the resulting dll is missing a lot of data because the original `jssc-2.8.0.jar` windows dll files were 66KB for the x86 version and 120KB for the x86_64 version and after I built them in VS they were 8KB and 9.5KB respectively. I tried inserting them into the .jar anyway but it the result was the same, the JVM crashes. Did your dll builds turn out any larger than mine? Would you be willing zip up your resulting dlls and/or the whole `jssc-2.8.0.jar` and providing those?

wholder, 2017-10-09:
I just replied to your post in the JSSC Wiki, but here what I said:

I only compiled the 64 bit DLL and, compiling it in debug mode, it came out
to be 81.5Kb. If you'd like to try my jssc.jar file, it's posted as part of
a project I have posted on Github. The links is:

  https://github.com/wholder/LaserCut

You'll find my modified jssc.jar file in the lib folder.

Wayne

astout, 2017-10-10:
@wholder you have saved my life! Your `.jar` file worked for my purposes. Thanks for your help!

tresf, 2019-07-24:
FYI, myself and a Java developer forked the JSSC codebase and released a build system as well as updated binaries.

Minus some MSVC compat items and scripting the version info into the header, the C code is identical.  The Java code has changed a bit to accommodate attempts to use a simpler technique to load the native libs.

Lastly, there are a few platforms that we've dropped support for (ARM soft float and legacy 32-bit/PPC mac being the most notable).

Due to some fundamental disagreements with the author of JSSC, our changes are to remain in a fork for the foreseeable future.  I wanted to share here for those that could benefit from it.

We've documented all of our tricks to get the `.dll` working on old Windows versions, etc and I'd be happy to share the techniques used (notably, optimization and static linking).

https://github.com/java-native/jssc/