arduino/Arduino #6789: Compilation error with version 1.8.4 of Arduino

darby59, 2017-10-03:
Hello,
I try to compile the blink exemple with the version 1.8.4 of arduino. It was working with the version 1.6.5, but now it give me this error : 
Arduino : 1.8.4 (Windows 7), Carte : "Arduino/Genuino Uno"

E:\arduino-1.8.4\arduino-builder -dump-prefs -logger=machine -hardware E:\arduino-1.8.4\hardware -tools E:\arduino-1.8.4\tools-builder -tools E:\arduino-1.8.4\hardware\tools\avr -built-in-libraries E:\arduino-1.8.4\libraries -libraries C:\Users\zfpydy\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923 -warnings=none -build-cache C:\Users\zfpydy\AppData\Local\Temp\arduino_cache_620471 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=E:\arduino-1.8.4\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=E:\arduino-1.8.4\hardware\tools\avr -prefs=runtime.tools.avrdude.path=E:\arduino-1.8.4\hardware\tools\avr -verbose E:\arduino-1.8.4\examples\01.Basics\Blink\Blink.ino
E:\arduino-1.8.4\arduino-builder -compile -logger=machine -hardware E:\arduino-1.8.4\hardware -tools E:\arduino-1.8.4\tools-builder -tools E:\arduino-1.8.4\hardware\tools\avr -built-in-libraries E:\arduino-1.8.4\libraries -libraries C:\Users\zfpydy\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923 -warnings=none -build-cache C:\Users\zfpydy\AppData\Local\Temp\arduino_cache_620471 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=E:\arduino-1.8.4\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=E:\arduino-1.8.4\hardware\tools\avr -prefs=runtime.tools.avrdude.path=E:\arduino-1.8.4\hardware\tools\avr -verbose E:\arduino-1.8.4\examples\01.Basics\Blink\Blink.ino
Using board 'uno' from platform in folder: E:\arduino-1.8.4\hardware\arduino\avr
Using core 'arduino' from platform in folder: E:\arduino-1.8.4\hardware\arduino\avr
Detecting libraries used...
"E:\arduino-1.8.4\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino-1.8.4\hardware\arduino\avr\cores\arduino" "-IE:\arduino-1.8.4\hardware\arduino\avr\variants\standard" "C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"E:\arduino-1.8.4\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino-1.8.4\hardware\arduino\avr\cores\arduino" "-IE:\arduino-1.8.4\hardware\arduino\avr\variants\standard" "C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923\sketch\Blink.ino.cpp" -o "C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923\preproc\ctags_target_for_gcc_minus_e.cpp"
"E:\arduino-1.8.4\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\zfpydy\AppData\Local\Temp\arduino_build_223923\preproc\ctags_target_for_gcc_minus_e.cpp"
E:\arduino-1.8.4\arduino-builder a retourné 1'073'741'855
Erreur de compilation pour la carte Arduino/Genuino Uno

I have tried on 3 PC with the same configuration : Windows 7 64bits.

Thaks for your help.
Best regards.
Darby

matthijskooijman, 2017-10-03:
It's weird that there is no actual error, just "arduino-builder returned some error code". Is this just the unmodified blink example? Have you been using older version of Arduino on these computers before? Could you try the latest version (1.8.5, released this week).

facchinm, 2017-10-05:
It looks like `ctags` is failing in creative ways (could be a dll error but it's impossibile since it's compiled statically). The error code is https://stackoverflow.com/a/4532492 and the issue seems really difficult to reproduce (after a bit of googling). Leaving it open in case some more details appear.

facchinm, 2017-10-05:
Probably it's worth moving the issue to https://github.com/arduino/arduino-builder . 
@darby59 would you mind reopening it there? Many thanks!

darby59, 2017-10-05:
The problem is coming with Impero service. If I stop it, the compilation is OK.

closed 2017-10-05 by darby59