arduino/Arduino #6790: Arduino Leonardo ETH -> SCHEMATICS link loop

Serg-L, 2017-10-03:
Page   **https://store.arduino.cc/usa/arduino-leonardo-eth**    link    <**SCHEMATICS**> 
is forwarding to page    
 **http://download.arduino.org/products/LEONARDOETH/Arduino_Leonardo_Eth_V1_sch.pdf**
Which is is forfarding back to:  **https://www.arduino.cc/**

Loop...

per1234, 2017-10-03:
Thanks for reporting the broken link. It looks like it was not updated when that page was moved from arduino.org.

As a workaround, you can access that file here:
https://www.arduino.cc/en/uploads/Main/Arduino_Leonardo_Eth_V1_sch.pdf

Note that this schematic is for v1 of the board and apparently there are some differences in the v2 board (https://github.com/arduino/Arduino/issues/6741).

per1234, 2017-10-10:
@frrederik the links still point to .org if you set your locale to "America, Asia, or Oceania".

kengdahl, 2017-12-27:
The loop is fixed now. The schematics are now downloadable from the US page.

closed 2017-12-27 by kengdahl