arduino/Arduino #6792: Please, add library to Arduino Library Manager

fu-hsi, 2017-10-03:
Hi.

My library:
https://github.com/fu-hsi/PMS/

Don't know why was not found. Now I have tags 1.0.0 and v1.0.0 without success. Which is correct?

Regards.

per1234, 2017-10-03:
I believe that Library Manager is smart enough to work with either release version format but it's best to use the version that exactly matches the `version` value in your library.properties file.

cmaglie will add you library to the Library Manager index on his next run through all the outstanding inclusion requests.

fu-hsi, 2017-10-04:
OK.
I deleted tag "v1.0.0" and leaved "1.0.0" to avoid possible duplicates.
For now, no progress.
From site https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ:
> The job runs every hour. If a new library has been released, you can expect it to be listed within the hour.

per1234, 2017-10-04:
Your library must be added to the index first before it will be available for installation via Library Manager. cmaglie occasionally comes through and adds all the outstanding requests to the index and then closes the issues at that time. Please be patient. You will get a notification when the issue is closed and then you can expect your library to appear in Library Manager shortly after that has been done.

cmaglie, 2017-10-05:
Done!

closed 2017-10-05 by cmaglie