arduino/Arduino #6800: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/optiboot/optiboot_atmega328-Mini.hex is "stale."

WestfW, 2017-10-07:
This .hex file seems to be from a VERY old version of optiboot, that seems to have been left behind back when the first Uno patches were made (~2012)  It's not a target that is supported in the current makefiles, doesn't contain a version number, has bugs that have long since been fixed, and thus is occasionally causing confusion. ( https://github.com/Optiboot/optiboot/issues/219 )

Boards.txt still points at this, but it's not clear how it's different than the normal optiboot .hex file (aside from being old and buggy.  Different or no "startup blink", perhaps?)  Otherwise I'd recommend deleting it entirely.  The boards.txt file also has an incorrect value for the mini upload size using this bootloader.

uski, 2017-11-03:
I agree this is annoying. For instance, recent versions of optiboot allow using the watchdog properly, whereas the default bootloader (in 1.8.5 at least) for atmega328p chips will be stuck in a reboot loop if a WDT reset happens.

I know a lot of people do not use the watchdog timer in their arduino sketches, but this may be one of the reason. The version of optiboot provided with the Arduino environment cannot even be compiled with the supplied compiler due to the -mshort-calls compiler flag no longer being supported.

The optiboot component of Arduino is long due for a refresh !

WestfW, 2017-11-03:
> recent versions of optiboot allow using the watchdog properly, whereas the default bootloader (in 1.8.5 at least) 
Really?   The Optiboot WDT fix is supposed to have gone in back around 2011, and is in the source code, and SHOULD be in all of the .hex files as well (around the 1.0 timeframe) ( https://forum.arduino.cc/index.php?topic=28733.0 ) This bug exclusively complains that the 328-mini.hex file does NOT match the current source.

Optiboot HAS had a lot of updates in various places (mostly feature additions, new chip support, and new compiler version workarounds), and it MIGHT be time for a refresh, but I'm not actually aware of any bugs that NEED an update.

ortegafernando, 2018-09-17:
Hi, any plan to update the optiboot that is installed with Arduino IDE ?