arduino/Arduino #6802: Saved window location on single monitor

voji, 2017-10-08:
Mostly I using arduino ide on my laptop with external monitors (desktop setup). Yesterday I detach my laptop, and plan to flash one of my board on external location. 

Starting Arduino IDE (latest) but the window not show in my laptop monitor. 

To fix this issue, I need to modify c:\Users\[user]\AppData\Local\Arduino15\preferences.txt file and delete  all ".*\.location=" property, and after that, the app open in my primary monitor. 

Possible solutions of this problem if on application start the ide check actual desktop size, and if the saved locations outside with the active desktop area, ignore the saved coordinates.

facchinm, 2017-10-09:
Hi @voji ,
which version of the IDE are you using? This should be solved in both 1.8.4 and 1.8.5 with https://github.com/arduino/Arduino/commit/8292989e2dab39c62d03ddd9c25b7f4570813b7d . If you are using one of these versions I'll take a closer look

voji, 2017-10-09:
I tested on 1.8.2, now upgraded to 1.8.5, and the issue still exists.

Steps: 
- Extend desktop to two monitor (or up)
- Open sketch, and move window away from primary monitor
- Close sketch
- Change monitor settings to use only the primary monitor
- Open arduino ide

Result: 
The ide started, icon show in the taskbar, but the window opened in the previously saved location, outside the active dekstop area. 

You can reproduce the problem with resolution change on a single monitor too. Set monitor to high res (for example 1920x1080), move the Arduino IDE window to upper right corner of a screen, and change the window size something small (for example, 200x200 px). Now exit the ide, change the resolution to 640x480, and reopen the IDE.

facchinm, 2017-10-09:
I tested it on a Win 10 VM (enabling and disabling the secondary monitor) and I'm currently usable to reproduce the problem. Every time I change the window setting and reopen the IDE it reverts to the center of the primary screen.
I'll test on a real PC tomorrow and report back if I'm able to reproduce it.

fortunereject, 2018-04-17:
The problem still exists in 1.8.5, confirmed.