arduino/Arduino #6804: a specific IDE border color IDE Arduino for each serial port used

okpapt, 2017-10-08:
Hello,
Is it possible having / choosing a specific border color for each IDE Arduino opened with specific serial port associed ?
Of course, if several windows IDE Arduino are opened and having same serial port, they have same border color.
Idem for serial monitor.

If it is impossible to define this in style file, maybe I can compil an IDE_Arduino_red, an IDE_Arduino_yellow, an IDE_Arduino_green,  etc. 

If it is not too dificult, can you explain me how can I do this ?
Thank you very much.

per1234, 2017-10-08:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2017-10-08 by per1234

okpapt, 2017-10-12:
It is a feature requests not a bug.

it is not within the competences  http://forum.arduino.cc/index.php?topic=148850

per1234, 2017-10-12:
>It is a feature requests not a bug.

Sorry, it sounded to me like a support request rather than a feature request. In that case I would have reopened this but you already created a duplicate issue https://github.com/arduino/Arduino/issues/6819 so I'll leave this one closed.

>It is not within the competences of Arduino forum.
http://forum.arduino.cc/index.php?topic=148850

How would you know it's not "within the competences" without posting about this on the forum? In fact if you would do so I will instruct you there how to edit the Arduino IDE theme so that you could achieve your desired workaround of having one color for each of multiple IDE installations.