arduino/Arduino #6805: Please add UltraDistSensor library

shubhamtivedi95, 2017-10-08:
the library is available here https://github.com/shubhamtivedi95/UltraDistSensor

per1234, 2017-10-08:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

shubhamtivedi95, 2017-10-09:
yes i finishes the task you can check

shubhamtivedi95, 2017-10-09:
yes i have finished task you can check

shubhamtrivedi95, 2017-10-09:
[image]  my library is not available on software can you please solve this issue.

per1234, 2017-10-09:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through and clear all the outstanding Library Manager inclusion requests and closes the issues. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2017-10-12:
Done!

closed 2017-10-12 by cmaglie