arduino/Arduino #6806: Keyboard shortcut for Increase Font Size not working

AlfGit, 2017-10-08:
Edit menu shows Increase Font Size shortcut as 'Ctrl+ Unknown KeyCode: 0x2b'. The same behavior is seen on W10 and W7, with Logitech USB (HID keyboard device) and Microsoft PS/2 (standard PS/2 keyboard) keyboards.

per1234, 2017-10-08:
[image]

AlfGit, 2017-10-08:
Fair enough, although I think one is a consequence of the other. It appears that KeyStroke.getKeyStroke does not like '+' as an argument. To clarify, it is not only displayed incorrectly, but the shortcut Ctrl+ Plus does not work.

per1234, 2017-10-09:
Sorry, I misunderstood the issue report because you only talked about the menu issue in the description and I tested that the new keyboard shortcut Ctrl+MouseWheel was working correctly. I forgot it is also Ctrl+Plus, though I should have noticed that Ctrl+Minus is Decrease Font Size. I've updated the title. Thanks for the issue report!

AlfGit, 2017-10-09:
Thank you. That describes the issue perfectly.

facchinm, 2017-10-09:
I took a look at this issue and it seems quite complicated to solve.
In fact, on keyboards with english layout the `+` key is only reachable through a modifier, so it breaks the shortcut. A solution would be to use the `+` key on the NumPad but it doesn't exist either on most laptops keyboards. If we switch to `=` which is primary on en keyboards we lose the integration for a lot of other locales...
I'll think about it and report back if I find something interesting/feasible.

AlfGit, 2017-10-09:
The + and - on the numpad are already assigned to opening and folding code blocks, which I would say is far more useful because it's a function that would likely be used repeatedly. I imagine that most people would set their preferred font size once and then leave it, and there is still Ctrl+ mousewheel for those of us that use zoom more often. Of course, not all users have mousewheel functionality either, especially those using a trackpad on a laptop.

Bradfo69, 2017-10-24:
Try to change your [keyboard](https://www.icrfq.com/family/314). A few months ago I got the same issues and I'm trying too may way to solve this but I can't.

ElectricRCAircraftGuy, 2018-01-29:
This problem still exists...
Running a Dell Precision 7510 with Ubuntu 14.04

facchinm, 2018-01-30:
@ElectricRCAircraftGuy would you mind testing the beta? It contains https://github.com/arduino/Arduino/commit/3d5c1bdba6bd08d45d70ea1d7b22fbc23610388f which should solve the issue for different keyboard layouts.

ElectricRCAircraftGuy, 2018-01-30:
Yeah...but I've never used the Beta. Do I need to build it or are binaries available?

ElectricRCAircraftGuy, 2018-01-30:
Nevermind, getting it now off the "beta builds" section here: https://www.arduino.cc/en/Main/Software

ElectricRCAircraftGuy, 2018-01-30:
I tested 1.9.0-beta and the menu looks better (no longer has that weird error), but Ctrl + PLus still does not work. Ctrl + Minus is fine

per1234, 2018-01-30:
The Ctrl+Plus keyboard shortcut (by "Plus" I mean the =/+ key) does not work for me with Arduino IDE 1.9.0-beta build 36 on Windows 10.

ReAlUA, 2018-01-30:
The same for 64-bit Ubuntu. Ctrl+Plus displayed correctly in menu but does not work.

Jim-2249, 2018-02-22:
With my setup, Win 10 & 1.8.5, ctrl+wheel either way just reduces the font size.

How about disabling the wheel for font size changes? I accidentally reduce the font with the wheel then have to go to preferences to fix it. If ctrl+wheel didn't change the font size, this issue would be moot for me.

NathanielJS1541, 2018-04-22:
I have the same problem. Also for me zooming out with the mouse wheel works, but zooming in does not.
[image]

Also I found it interesting that no matter which way I scroll, it always zooms out.

I have also tested this on the laptop touchpad (using two fingers on the touchpad to scroll up or down), and I get the same result.

JayConverse, 2018-06-25:
Ditto here, Windows 10 Dell, Arduino 1.8.5, Teensyduino 1.41

PaulStoffregen, 2018-06-25:
Mouse wheel while holding CTRL is able to scroll in and out when I try with Arduino 1.8.5 on Windows 10 and Ubuntu 14.04.

Here's a quick screen recording from Windows 10.

https://youtu.be/-z6nPbKy6Es

JayConverse, 2018-06-25:
Thanks, Paul!  I knew that wheel thing was useful for something....

Jim-2249, 2018-06-26:
I had tried and found ctrl+wheel only decreased font size. Just tried it again and it does both. What changed?

PaulStoffregen, 2018-06-26:
I spent a couple hours looked into this bug (the keyboard shortcut, not scroll wheel).

The fix in 1.9 to use KeyEvent.VK_PLUS does **not** work on US keyboards, and presumably all others where '+' is typed with the shift key.

Sadly, it seems all roads not from actual keyboard input lead to "KEY_TYPED" codes & events, which don't have any awareness of the keyboard layout.  Java's KeyEvent.getExtendedKeyCodeForChar() looked like the most promising option, but it turned out to be an empty promise.  It returns VK_PLUS, even when a US keyboard is used.

Digging deeper into this, I found many references to the keyboard layout being handled by the host system.  After a couple hours, the reality started to sink in that Java doesn't seem to actually have any logic/knowledge of the keyboard layout at all.  :-(

As nearly as I can tell, the *only* solution is to query the Locale info.  But even that appears to be less ideal.  It also seems to lack any solid info about the actual keyboard layout used.  Whether to use VK_PLUS or VK_EQUALS can be clear in many common cases, but there doesn't seem to be any Java way to perfectly find this info.

pendletonrc, 2018-07-12:
May I suggest that you simply get rid of the the keyboard shortcuts for changing font size? You provide the wheel based font zoom and you can set both the font size and GUI scale factor in the preferences menu.
Old saying from way back... when in doubt, cut it out.

Jim-2249, 2018-07-12:
With 1.9.0 beta build 71, sometimes ctrl+scroll-up increases font size and sometimes it does not. I think I try it just after starting the program. Also, ctrl+minus works and ctrl+plus does not at the moment.

cmaglie, 2018-08-15:
It seems that most keyboards have either "+" directly available on the base layer or as SHIFT+"=", so the simplest solution is to use two keybinding for the "Increase Font Size" action:
"CTRL +" and "SHIFT CTRL =" (to allow keyboards with "+ =" on the same key).

See #7888

commit 2988136f (2018-08-20): Added alternative key binding for "Increase Font Size"

This commits adds the "CTRL SHIFT =" as key to increase font size.
The old shortcut "CTRL +" is still active, but it's not always
possible to produce it if the keyboard layout doesn't have the + key
on the base layer.

This add the possibility to compose "CTRL +" when the keyboard has
the "+" available as "SHIFT =", that seems to be very common in
many layouts.

Fix #6806
closed 2018-08-20 by facchinm