arduino/Arduino #6809: Very slow to open or save in 1.8.4 when connected to wireless

spacelego, 2017-10-10:
When the PC is connected to a wireless LAN, Arduino IDE is VERY slow to open, and if you try to save or compile a sketch, it also hangs.

I suspect this has to do with searching? for a UnoWifi board via some kind of discovery.  Is there no way to specify the IP of the board(s) you want to connect, to instead of waiting for the discovery or for it to timeout and find nothing?

facchinm, 2017-10-10:
Could be a duplicated of https://github.com/arduino/Arduino/issues/6350, we are working on a long term solution to solve it once for all.
The hang while compiling, instead, could be due to an antivirus checking all the intermediate files produced by the build. If you are on windows, try pausing your AV and check if it alleviates the issue.

spacelego, 2017-10-10:
I took a look at thread 6350, and while I do also have the VirtualBox Host-Only Network "NIC" that does not seem to be related.  Tested on two different machines...

Again, when connected to the LAN with WiFi, opening the IDE takes forever, a save takes forever, a compile or an upload takes forever.
On the same machine, when disconnected from wireless, but connected to the LAN with a hard-line Ethernet cable, opening the IDE is fast, as is save/compile/upload.

There is NO anti-virus anti-malware product installed on either machine.