arduino/Arduino #6811: unable to upload to the board

aprilmintacpineda, 2017-10-10:
About 3 weeks ago I bought an arduino uno board from [Lazada](http://www.lazada.com.ph/uno-r3-ch340g-mega328p-for-uno-r3-with-usb-cable-36576502.html?spm=a2o4l.020606000000.0.0.8BkA10&cid=1804&did=4aec79eb-bf5f-48f8-b991-10aafba221df&mp=1&impsrc=OE721ELAALRYLYANPH-43702041), it turns out that the board is actually a clone according to [this guy](http://www.instructables.com/id/How-to-fix-bad-Chinese-Arduino-clones/). It actually worked like the first and second day I used it. Now I can't upload any program to the board, when I upload a program the `arduino IDE` says

```
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aprilpineda\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805 -warnings=all -build-cache C:\Users\APRILP~1\AppData\Local\Temp\arduino_cache_175320 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\examples\11.ArduinoISP\ArduinoISP\ArduinoISP.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aprilpineda\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805 -warnings=all -build-cache C:\Users\APRILP~1\AppData\Local\Temp\arduino_cache_175320 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\examples\11.ArduinoISP\ArduinoISP\ArduinoISP.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp" -o "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp" -o "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp.o"
Compiling libraries...
Compiling library "SPI"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\libraries\SPI\SPI.cpp.o"
Compiling core...
Using precompiled core
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/ArduinoISP.ino.elf" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\sketch\ArduinoISP.ino.cpp.o" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805\libraries\SPI\SPI.cpp.o" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/..\arduino_cache_175320\core\core_arduino_avr_uno_0c812875ac70eb4a9b385d8fb077f54c.a" "-LC:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805" -lm
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/ArduinoISP.ino.elf" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/ArduinoISP.ino.eep"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -R .eeprom  "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/ArduinoISP.ino.elf" "C:\Users\APRILP~1\AppData\Local\Temp\arduino_build_337805/ArduinoISP.ino.hex"
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Sketch uses 4420 bytes (13%) of program storage space. Maximum is 32256 bytes.
Global variables use 483 bytes (23%) of dynamic memory, leaving 1565 bytes for local variables. Maximum is 2048 bytes.
[... cut ...]
         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

Tools->board = Arduino/Genuino Uno
Tools->port = COM7 // it's the only port there,
Tools->programmer = AVRISP mkll

Tools->Board info says:

```
BN: Unknown board
VID: 1A86
PID: 7523
SN: Upload any sketch to obtain it
```

Arduino IDE 1.8.5

aprilmintacpineda, 2017-10-10:
Please also note that:

I was able to upload my programs the past days so there's still a program in the board, and when I open the serial monitor, the program actually runs, I use finger print scanner with it.

aprilmintacpineda, 2017-10-10:
I have fixed it, closing this.

closed 2017-10-10 by aprilmintacpineda

evangael, 2017-12-10:
How did you fix it by the way? I have the same issue and didn't find a correct answer yet on how to fix it. :/

7043mcgeep, 2018-03-22:
@aprilmintacpineda Mind describing your fix?

aprilmintacpineda, 2018-03-23:
@evangael  @7043mcgeep sorry for a late response. Turns out that there was actually no problem, the only problem was that I was pressing the wrong buttons. I use keyboard shortcuts to upload to my Arduino board and I was pressing `Shift+CMD+U` or `Shift+CTRL+U` on windows. It should have been `CMD+U` or `CTRL+U` on windows.

7043mcgeep, 2018-03-23:
@evangael @aprilmintacpineda Thanks for your reply. I actually fixed my issue as well. I did not immediately realize that the TX/RX pins need to be unoccupied in order for an upload to transfer. I was actually using TX/RX instead of a software serial, so there lies my issue.