arduino/Arduino #6812: No indent for namespace with auto format

SUPERETDUPER, 2017-10-10:
```
namespace {
   //Some code
}
```
after running auto format (Ctrl+T) becomes
```
namespace{
//Some code
}
```

To me, this seems like incorrect formatting.

per1234, 2017-10-10:
Just a note that it is possible to customize the auto format configuration:
1. Open {Arduino IDE installation folder}/lib/formatter.conf in a text editor.
2. Add the line:
```
indent-namespaces
```
3. Save the file to the folder of the link shown after **File > Preferences > More preferences can be edited directly in the file**
4. Restart the Arduino IDE.

You can find documentation of all the configuration options here:
http://astyle.sourceforge.net/astyle.html

That's not to say that it's not reasonable to request that this be set by default, but only to make it clear that we do each have the option to configure the formatter to our own preferences if the defaults are not to our liking.

SUPERETDUPER, 2017-10-10:
@per1234 Thank you, that worked. If you want this to be the default, and point me in the right direction I can start working on a pull request...

per1234, 2017-10-10:
At a glance it does seem sensible to indent.

I tend to just accept whatever style conventions were originally set for the formatter but I don't know whether that option was intentionally left out or just never considered. It's somewhat unfortunate to change the established standards but I don't think this change would have a huge impact since it's not very common to see namespaces used in Arduino sketches. It's certainly better to make the change now rather than later if it's going to be done.

If you want to submit a pull request the file is here: https://github.com/arduino/Arduino/blob/master/build/shared/lib/formatter.conf

cmaglie, 2017-10-30:
As said in #6813 we will never find a configuration that match everyone's personal taste, so I'll close this as wontfix and just keep the existing configuration.

closed 2017-10-30 by cmaglie