arduino/Arduino #6814: Icon files for Arduino IDE

navy57, 2017-10-11:
If you want to change the icon for Arduino files, here is the file:
Multi-sizes. 
[arduino_ico_ino_pde.zip](https://github.com/arduino/Arduino/files/1373849/arduino_ico_ino_pde.zip)

navy57, 2017-10-11:
Add the icon file to the default installation folder Arduino:
"C:\Program Files (x86)\Arduino"
  and run the following reg file:
[Arduino reg_master.zip](https://github.com/arduino/Arduino/files/1373892/Arduino.reg_master.zip)

per1234, 2017-10-11:
This issue tracker is only to be used for reporting bugs or feature requests. If I understand correctly, this is neither so I'm closing as invalid. In the future please post things like this to the [Arduino Forum](http://forum.arduino.cc/) or the [Arduino Playground](http://playground.arduino.cc/) wiki.

closed 2017-10-11 by per1234