arduino/Arduino #6815: [IDE] No "Increase Font Size"/"Decrease Font Size" in .po files

ReAlUA, 2017-10-11:
Strings "Increase Font Size" and "Decrease Font Size" ([Editor.java](https://github.com/arduino/Arduino/blob/ba04ed2f7fa7f1c2733eb77a784253cf58365e82/app/src/processing/app/Editor.java#L1378)) not exist in [`.po` files](https://github.com/arduino/Arduino/tree/master/arduino-core/src/processing/app/i18n) so they are can't be translated on [transifex](https://www.transifex.com/mbanzi/arduino-ide-15/).

Looks like some script must be launched for updating `.po` files here and on IDE translation project.

facchinm, 2017-10-11:
They should be, since they are wrapped in a `tr()` (see https://github.com/arduino/Arduino/blob/ba04ed2f7fa7f1c2733eb77a784253cf58365e82/app/src/processing/app/Editor.java#L1378) .
@cmaglie is there any special procedure we should follow to tell Transifex to pick up that strings?

feikname, 2017-10-11:
@facchinm @ReAlUA 

The README file present in [`arduino-core/src/processing/app/i18n`](https://github.com/arduino/Arduino/tree/master/arduino-core/src/processing/app/i18n) has the answers for your questions.

ReAlUA, 2017-10-11:
@feikname
Is any Transifex user can push to Arduino IDE 1.5 project? 
Anyway, I can't make changes in arduino/Arduino repository but only in forked one. I think it is not a good idea to do this job via fork & pull request.

closed 2017-10-13 by cmaglie

cmaglie, 2017-10-13:
A change in the "translatable" strings in the code is propagated to transifex in the following way:

1) the script `arduino-core/src/processing/app/i18n/update.sh en` must be run (with the parameter `en`), to update the files `Resources_en.po` and `Resources_en.properties`. All the new/changed strings are loaded into those files and they can be commited together with the patch (if the author knows how to do it and he/she remembers to do it).

2) An admin of the transifex project (me in this case), picks the new `Resource_en.po`, that is the **base language**, and update the resource on transifex. This action will update all the strings to be translated on the other languages.

This is enough to allow translators to work on new strings.

3) Before a release, I run the `pull.sh` with `-a` to donwload from transifex all the language resources that have been updated by translators in the meantime so they can be included in the next release.

ReAlUA, 2017-10-13:
Thanks! I just made a translation of new lines for the Ukrainian on Transifex :-)

commit 81997612 (2017-10-13): Updated source translations

Fix #6815