arduino/Arduino #6816: Error compiling for board Arduino/Genuino Mega or Mega 2560.

gril002, 2017-10-11:
Arduino IDE 1.9 fails to compile even a blank sketch for the Mega 2560. It does not provide nay reason why only exit status.

gril002, 2017-10-11:
[image]

gril002, 2017-10-11:
Update: it did not compile due to the fact that Windows defender ha quarantined arduinoOTA.exe due to being detected as a Trojan.
[image]
[image]

gril002, 2017-10-11:
It appears that even after restoring it, it still doesn't want to compile for any Arduino...

facchinm, 2017-10-11:
Hi @gril002 , could you paste the output when launching it with `arduino_debug.exe` ? thanks!

gril002, 2017-10-11:
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
Preparing boards...
Starting...

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

facchinm, 2017-10-11:
That's kinda strange... Can you also enable verbose output from Preferences? At leas one line calling `arduino-builder.exe` should be displayed and we can continue the debugging from there.
Thanks for testing btw!

gril002, 2017-10-11:
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
Preparing boards...
Starting...

C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware -hardware C:\Users\Gašper Gril\AppData\Local\Arduino15\packages -tools C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\tools-builder -tools C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -tools C:\Users\Gašper Gril\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\libraries -libraries C:\Users\Gašper Gril\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10900 -build-path C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774 -warnings=none -build-cache C:\Users\GAPERG~1\AppData\Local\Temp\arduino_cache_151382 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -verbose C:\Users\GAPERG~1\AppData\Local\Temp\untitled313175031.tmp\sketch_oct11a\sketch_oct11a.ino
C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\arduino-builder -compile -logger=machine -hardware C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware -hardware C:\Users\Gašper Gril\AppData\Local\Arduino15\packages -tools C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\tools-builder -tools C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -tools C:\Users\Gašper Gril\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\libraries -libraries C:\Users\Gašper Gril\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10900 -build-path C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774 -warnings=none -build-cache C:\Users\GAPERG~1\AppData\Local\Temp\arduino_cache_151382 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr -verbose C:\Users\GAPERG~1\AppData\Local\Temp\untitled313175031.tmp\sketch_oct11a\sketch_oct11a.ino
Using board 'mega' from platform in folder: C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Users\Gašper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr
Detecting libraries used...
"C:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr\cores\arduino" "-IC:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr\variants\mega" "C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774\sketch\sketch_oct11a.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr\cores\arduino" "-IC:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\hardware\arduino\avr\variants\mega" "C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774\sketch\sketch_oct11a.ino.cpp" -o "C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Users\GaĹˇper Gril\Desktop\arduino-PR-beta1.9-BUILD-44\tools-builder\arduino-preprocessor\0.1.3/arduino-preprocessor" "C:\Users\GAPERG~1\AppData\Local\Temp\arduino_build_593774\preproc\ctags_target_for_gcc_minus_e.cpp" "" -- -std=gnu++11
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

gril002, 2017-10-11:
Not sure what happened to the text....

gril002, 2017-10-11:
If it helps
[Log.txt](https://github.com/arduino/Arduino/files/1376908/Log.txt)

facchinm, 2017-10-11:
Great, thanks a lot! It looks like the new preprocessor is crashing on the path name; I believe it's a unicode problem. One last thing, can you paste your Windows username as it is correctly displayed by the OS (I believe it's `Gašper ` but I want to be 100% sure).

gril002, 2017-10-11:
Yeah you're correct it's C:\Users\Gašper Gril

facchinm, 2017-10-13:
Hi @gril002 ,
the[ new build ](https://www.arduino.cc/download_handler.php?f=/ide_beta/arduino-1.9.0-beta-windows.zip)should solve both your issues (compilation failure and no autocomplete).
Could you test it? Thanks a lot!
PS: the build name is **PR-beta1.9-BUILD-10**

gril002, 2017-10-13:
It seems the weird charracters are still there but it does work now so it's fine
PS: Is it normal for it to spam out using proxy direct every so often? I don't remember it happening in 1.8
[Log.txt](https://github.com/arduino/Arduino/files/1383192/Log.txt)

gril002, 2017-10-13:
Oh and autocomplete also works fine now

closed 2018-05-31 by cmaglie