arduino/Arduino #6817: Please add Polymorphic Buttons Library

JCWentzel, 2017-10-11:
https://github.com/JCWentzel/PolymorphicButtons

per1234, 2017-10-11:
In order to be added to the Library Manager index, your library must be located in the root of the repository. Currently your library is in a subfolder named PMButton. Please move the library to the root of the repository and then re-do or create a new release/tag.

After you have done so, please comment here to let us know we can proceed.

More information: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

JCWentzel, 2017-10-12:
The Library Has now been moved to the root and a new release has been made.

per1234, 2017-10-12:
The `version` value in your library.properties must match the version of the release. Your release is 1.0.2 but the `version` is still set to [1.0.1](https://github.com/JCWentzel/PolymorphicButtons/blob/1.0.2/library.properties#L2) in library.properties. `version` must be updated before the release is created so that the release will contain the library.properties file with the correct value.

Please comment here once this is resolved.

JCWentzel, 2017-10-12:
Version mismatch has been resolved.

cmaglie, 2017-10-12:
Done!

closed 2017-10-12 by cmaglie