arduino/Arduino #6818: Arduino IDE 1.9 Beta : com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input

erkobg, 2017-10-11:
Downloaded **arduino-PR-beta1.9-BUILD-44**

After pressing Autocomplete  ( Ctrl + Space) for the first time after a while I see exception :

`com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input
 at [Source: ; line: 1, column: 1]
	at com.fasterxml.jackson.databind.JsonMappingException.from(JsonMappingException.java:148)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3781)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2726)
	at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:98)
	at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
	at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
	at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1663)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6310)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:806)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1074)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:945)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:771)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

erkobg, 2017-10-11:
#

facchinm, 2017-10-12:
Hi @erkobg ,
a new build of beta branch is live on the downloads server; it should solve your issue and add a refined diagnostic on preprocessor/json related errors.
The new version name is `arduino-PR-beta1.9-BUILD-9` (we had a small problem with Jenkins, they'll all be sorted from now on).

Thanks for testing! Feel free to report if it fixes your problem.

erkobg, 2017-10-12:
Hello @facchinm,
downloaded the latest build available: **arduino-PR-beta1.9-BUILD-9**

Exception is gone, but nothing happens else....
Well when I press for Autocomplete it hangs for a long period( that's normal) but without any result.
I expect to see autocomplete list ( for example I hit Autocomplete on Serial. and nothing happens).
If I hit Autocomplete - again hanging and no result...

I started Arduino from debug : arduino_debug.exe and the output is following:

`_Looking for library user32
Adding paths from jna.library.path: null
Trying user32.dll
Found library user32 at user32.dll
Looking for library shell32
Adding paths from jna.library.path: null
Trying shell32.dll
Found library shell32 at shell32.dll
Looking for library Ole32
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library Ole32 at Ole32.dll
Loading configuration...
Initializing packages...
Preparing boards...
Starting...
"RCSwitch.cpp" contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.

"RCSwitch.h" contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.

Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT_`

Please advice.

facchinm, 2017-10-13:
Thanks for testing, let's try debugging this.
Which board have you currently selected?
Two things you could try:
- selecting an Arduino Uno (and try autocompleting Serial-related stuff)
- moving `RCSwitch` library outside the sketchbook (after closing the IDE and killing arduino-builder.exe process)
In case of error the new build should report (in verbose mode) a string to the commandline which is being launched.

erkobg, 2017-10-13:
Hello @facchinm,
I was trying my project for NodeMCU, so after your suggestions I tried :
1. Trying smaller project as Blink- it is not working for NodeMCU.
2. Trying smaller project as Blink- **it is  working for Arduino Uno**.
13  For my project, changing to Arduino Uno board - autocomplete works

So at least my conclusion is that Autocomplete is working for default boards only.

facchinm, 2017-10-13:
Yep, there could be some edge cases we are still trying to figure out; we currently preprocess the source (with gcc or any default compiler for the target) before passing it to the preprocessor.
This means that there **could** be gcc-specific stuff that gets imported only if used (so not in every sketch or board combination) and "breaks" the (clang based) preprocessor. 
We will add a bit of debugging to understand how the AST generation is failing in these cases.

facchinm, 2017-10-13:
Could you test the latest build (10) after selecting nodemcu? We silenced a couple of warnings with could have caused the error.

erkobg, 2017-10-13:
Hello @facchinm,
Seems that this build works even for the NodeMcu.

The only annoying thing left - every time on Autocomplete it freezes and takes quite some time to show the list, not only the first time. Time for showing the list is proportional to the amount of the code in the sketch - if the code is bigger it takes more time.

fizban99, 2017-12-27:
Hi, I just tested with TinyScreen+ with the current beta (build-31) and I am getting a similar problem. CircuitPlaygroundExpress shows autocompletion fine.
In debug window the TinyScreen+ default example displays the following when triggering autocomplete after `display.`:
`
com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input
 at [Source: ; line: 1, column: 1]
        at com.fasterxml.jackson.databind.JsonMappingException.from(JsonMappingException.java:148)
        at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3781)
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2726)
        at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:99)
        at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
        at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
        at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
        at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1663)
        at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
        at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
        at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
        at java.awt.Component.processEvent(Component.java:6310)
        at java.awt.Container.processEvent(Container.java:2236)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
        at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:806)
        at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1074)
        at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:945)
        at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:771)
        at java.awt.Component.dispatchEventImpl(Component.java:4760)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Using proxy DIRECT
Using proxy DIRECT`

jbrepogmailcom, 2019-11-26:
Hello, got this error with file arduino-1.9.0-beta-linux64  on Linux. So it is still not resolved?

com.fasterxml.jackson.databind.exc.MismatchedInputException: No content to map due to end-of-input
 at [Source: (String)""; line: 1, column: 0]
	at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4133)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3988)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2992)
	at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:99)
	at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
	at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
	at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:835)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:974)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:800)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)