arduino/Arduino #6819: feature requests : change border color IDE arduino

okpapt, 2017-10-12:
Hello,

It is not within the competences of Arduino forum.
 http://forum.arduino.cc/index.php?topic=148850

When I have several Arduino conected by USB cable, it is not easy to choice quickly the good Arduino editor.
Is it possible having / choosing a specific border color for each IDE Arduino opened with specific serial port associed ?
Of course, if several windows IDE Arduino are opened and having same serial port, they have same border color.
Idem for serial monitor.

If it is not important for developper, maybe it is possible to define this in style file or maybe I can compil an IDE_Arduino_red, an IDE_Arduino_yellow, an IDE_Arduino_green, etc. How can I do it ? Please explain me.

I know this IDE Arduino come from Open Source and are used for other program, can you give a link about original project ?

Thank you very much.

facchinm, 2017-10-12:
Hi @okpapt ,
the serial port is shared among all IDE instances unless you open more than one instance (eg. with double click on .ino files). Anyway, implementing your request works particularly bad with boards that change port name during upload, since the serial port is automatically switched to the one that "reappears".

I'd leave the issue here, tagging it properly, but I don't think we'll work on it anytime soon.

PaulStoffregen, 2017-10-12:
sort of related to #6490

okpapt, 2017-10-12:
Thank you for your answer.
Sorry for my explanation error.
I complete it with more text.

Each time I open a new instance of IDE (click on arduino.exe) I can use a specific serial port with a different Arduino board.

If I open a new window with file/open or with double click on .ino files I have same serial port for this instance.

One instance of Arduino IDE can have several windows with same serial port.

Each instance of Arduino IDE can have separte serial port.

I want a specific color for each instance of IDE Arduino.

matthijskooijman, 2017-10-12:
@facchinm, I think you might be misunderstanding the request here. AFAIU, @okpapt indeed knows you can start the IDE multiple times to connect to different serial ports. I also do that sometimes, when I have a few Arduinos attached and I want to keep their serial output open side by side. However, among these windows, it is not always easy to quickly see which is which (you'd have to look at the COMx number and remember which is which).

I think @okpapt is requesting that each serial port gets a different color, which is displayed somewhere in the GUI (e.g. the border of the window). This means that a given serial port (and, at least on Windows where a given Arduino always gets the same COM port, a give Arduino) has a specific color, which makes it easier to quickly distinguish them.

PaulStoffregen, 2017-10-12:
FWIW, better support for using multiple boards is one of the most commonly requested features I hear for Teensy.  Admittedly, this is one of the places Teensy is weakest, because I use HID protocol instead of serial for uploading, so part of the process involves searching for the first available device because it's no a serial port.

I understand multiple board support might be a too-complex feature for Arduino.  I know the use of specific colors is a careful and deliberate "branding" choice for Arduino.  Still, I'm interested in whatever can be done to better support using 2 or more boards.

okpapt, 2017-10-12:
If you read this 
 http://forum.arduino.cc/index.php?topic=480263.msg3279281#msg3279281
you can sea It is not within the competences of Arduino forum.

Thank you for your help.

okpapt, 2021-02-02:
UP !
In picture I made quickly, I forget change name of sketch, sorry !