arduino/Arduino #6820: Pressing Alt underlines English capital letters even in localized words

gril002, 2017-10-12:
[image]
[image]

As the title suggests when you press Alt in the editor it underlines the English capital letters not the local ones. This is true in Arduino IDE 1.8.4 and 1.9

facchinm, 2017-10-17:
The effort is probably feasible but it involves preparing another bunch of (uncorrelated) translation for the mnemonics (if i understood well the content of this tutorial http://uic.sourceforge.net/tutorials/i18n/).
Adding "Help needed" tag.

facchinm, 2017-10-17:
Never mind, it appears that it's quite easy :smile: 

for example:
```patch
-    fileMenu.setMnemonic(KeyEvent.VK_F);
+    fileMenu.setMnemonic(trm("File"));
```

with `trm` defined as
```java
  public static int trm(String s) {
          try {
             if (i18n == null)
               return s.charAt(0);
             else
               return i18n.getString(s).charAt(0);
           } catch (MissingResourceException e) {
               return s.charAt(0);
          }
  }  
```

The Help tag is still there if someone wants to speedup the effort and prepare a PR with the converted mnemonics.