arduino/Arduino #6821: Serial.flush() hangs when using 2Mbps

wlong007, 2017-10-13:
When using a buad rate of 2Mbps, Serial.flush() hangs. This issue is fully discussed and solved here at http://forum.arduino.cc/index.php?topic=504753.0. The proposed solution is to switch lines 221 and 222 shown here https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp#L221-L222.

matthijskooijman, 2017-10-13:
I think this is a known issue which was also discussed at #3745 (though that was originally about another problem). The fix you propose will work, but introduces other problems. Could see if the PR at #3865 fixes the problem for you? I think it should.

Coding-Badly, 2017-10-15:
> The fix you propose will work, but introduces other problems.

Such as?

matthijskooijman, 2017-10-16:
See (the last part of) https://github.com/arduino/Arduino/issues/3745#issuecomment-142244433

johnholman, 2017-10-25:
Should be fixed by #6855

sandeepmistry, 2019-09-16:
I believe this is closed by https://github.com/arduino/Arduino/pull/6855.

Please let us know if the issue still occurs and we can re-open.

closed 2019-09-16 by sandeepmistry