arduino/Arduino #6827: When saving new tab .ino file closes if it has unsaved changes on 1.9.x beta build 10

jmusta, 2017-10-15:
When saving newly created tab while .ino file has modifications, the .ino ino file gets closed. The changes seem to be saved properly though. Tested on 1.9.x beta build 10 on Windows 10.

1. Open sketch
2. Modify the .ino file, Do not save the changes
3. Create new tab.
4. Click Save
5. .ino file is closed,

Following exception is displayed when I try to save the changes

> Could not createInput() for E:\Users\username\Documents\Arduino\sketch_oct15b\test.h
java.io.FileNotFoundException: E:\Users\username\Documents\Arduino\sketch_oct15b\test.h (Määritettyä tiedostoa ei löydy)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at processing.app.legacy.PApplet.createInput(PApplet.java:595)
	at processing.app.legacy.PApplet.loadStrings(PApplet.java:254)
	at processing.app.BaseNoGui.loadFile(BaseNoGui.java:655)
	at processing.app.SketchFile.load(SketchFile.java:256)
	at processing.app.EditorTab.reload(EditorTab.java:381)
	at processing.app.tools.WatchDir.lambda$processEvents$0(WatchDir.java:167)
	at java.util.ArrayList.forEach(ArrayList.java:1249)
	at java.util.Collections$UnmodifiableCollection.forEach(Collections.java:1080)
	at processing.app.tools.WatchDir.processEvents(WatchDir.java:167)
	at processing.app.Editor$55.run(Editor.java:2012)
	at java.lang.Thread.run(Thread.java:748)
Warning: Failed to reload file: "test.h"

jmusta, 2017-10-15:
As a follow-up since I suspect these both issues are related: When I delete a tab when neither sketch nor tab are modified I see the following exception:

> Exception in thread "Thread-31" java.util.ConcurrentModificationException
        at java.util.ArrayList.forEach(ArrayList.java:1252)
        at java.util.Collections$UnmodifiableCollection.forEach(Collections.java:1080)
        at processing.app.tools.WatchDir.processEvents(WatchDir.java:167)
        at processing.app.Editor$55.run(Editor.java:2012)
        at java.lang.Thread.run(Thread.java:748)

If the sketch has unmodified changes while deleting a tab, in addition to the exception occurring the sketch loses the unmodified changes

facchinm, 2017-10-16:
Hi @jmusta , the issues seems related with the new filewatcher; it should solve the problem that, if you modify a sketch out of the Java IDE, when you save it back the external modifications are being overwritten. It is heavily based on the editor window focus status but it was only tested extensively on Linux so there could be subtle differences on other OSes.

Which operating system are you running? Could you provide a screencast of what you are doing so I can reproduce it correctly (with all the focused/unfocused windows in the same place)? Thanks!

facchinm, 2017-10-16:
Hi @jmusta,
I patched the wrong behaviour in 499e353 and it should be available in beta channel right now.
Would you mind testing it? Thanks a lot!

jmusta, 2017-10-16:
I tested it with build 12. The issue with saving in the original issue submission seems to be fixed. The bug with deleting the file that I included in the follow-up comment seems to still be present.

facchinm, 2017-10-16:
Just pushed another fix about the sketch deletion, should be online right now. Thanks for testing!

jmusta, 2017-10-16:
**Saving a new tab:**
Seems like the original bug, or at least similar, is showing up again in build 14. When I save new tab while the .ino ino file has unsaved changes it does not close immediately. Instead it closes when I try to choose the .ino file into focus after saving new tab. Steps to reproduce:
   1. Open sketch
   2. Modify the .ino file. Do not save the changes
   2. Create new tab
   3. Choose the new tab.
   4. Save changes while the newly created tab is in focus
   5. Choose the .ino file
   6. .ino file is closed

It does not anymore show me any exception messages even in when running as debug. I rechecked with build 12 and in that saving the new tab while the sketch has unsaved changes behaves correctly.

**Deleting tab while .ino file has unasaved changes**
For the deletion I do not consistently get any exceptions anymore either, but now I noticed that if the .ino file has unsaved modifications when I delete a tab those modifications are reverted. I rechecked with build 12 and same behavior seems to be present in that. I just did not notice it earlier.

At some point when I was playing around with the save and delete I received the following exception:

> Exception in thread "AWT-EventQueue-0" java.lang.IndexOutOfBoundsException: Index: 1, Size: 1
        at java.util.ArrayList.rangeCheck(ArrayList.java:653)
        at java.util.ArrayList.get(ArrayList.java:429)
        at processing.app.Editor.getCurrentTab(Editor.java:1592)
        at processing.app.SketchController.handleDeleteCode(SketchController.java:234)
        at processing.app.EditorHeader$Actions.lambda$new$2(EditorHeader.java:101)
        at processing.app.helpers.SimpleAction.lambda$new$0(SimpleAction.java:97)
        at processing.app.helpers.SimpleAction.actionPerformed(SimpleAction.java:110)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
        at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
        at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
        at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
        at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
        at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
        at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
        at java.awt.Component.processMouseEvent(Component.java:6533)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6298)
        at java.awt.Container.processEvent(Container.java:2236)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
        at java.awt.Container.dispatchEventImpl(Container.java:2280)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Unfortunately I have not been able to reproduce the exception.

facchinm, 2017-10-17:
Thanks for the extensive report, the last commit had some unwanted side effects that I didn't consider properly. I'll be back when I have a proper solution. Thanks for testing!

facchinm, 2017-10-17:
I pushed a new commit which should solve all the issues without side effects (https://github.com/arduino/Arduino/commit/3a3b695d53f8fde16dee9129dc4a52a058c17770). Me and @cmaglie are preparing a bunch of automatic tests based on your findings to quickly spot regressions on the Watcher codebase.
Thanks again!

per1234, 2018-01-16:
@jmusta, did you ever get a chance to try the latest beta build? If so, did it resolve the issues you reported here?

jmusta, 2018-01-18:
Sorry, I have not tested it. I'm currently rather busy, but I can try to find time to do it next week/this weekend.

jmusta, 2018-01-29:
I tested this with build 36. The saving seems to be working fine and none of the files get closed with following steps:
1. Open sketch
2. Modify the .ino file. Do not save the changes
3. Create new tab
4. Choose the new tab.
5. Save changes while the newly created tab is in focus
6. Choose the .ino file

When I deleted a tab while neither the ino file or tab were unsaved I the tab was successfully deleted, but I got the following error message:

> Exception in thread "Thread-48" java.util.ConcurrentModificationException
	at java.util.ArrayList.forEach(ArrayList.java:1258)
	at java.util.Collections$UnmodifiableCollection.forEach(Collections.java:1080)
	at processing.app.tools.WatchDir.processEvents(WatchDir.java:168)
	at processing.app.Editor$55.run(Editor.java:2022)
	at java.lang.Thread.run(Thread.java:748)