arduino/Arduino #6832: mDNS packet flood destroys my local network due to jmdns 3.5.1

kurtgo, 2017-10-16:
My simple network of 6 OTA arduino nodes and a few open Arduino IDE windows causes a massive flood of mDNS traffic.

The jmdns java package has a rather large issue relating to broadcast mDNS packets.  It seems that each service that responds to the addServiceListener() for the _arduino._tcp.local query causes an extra mDNS broadcast to be sent by jmdns.  This has major scaling issues when there are many OTA clients listening on a single network. Each broadcast causes N responses and creates N more broadcasts. leading to a massive flow of broadcast packets and responses.

Secondly, the Arduino code restarts the jmdns query every 5 seconds.  If the above process takes over 5 seconds (like when 6 or more OTA clients exist), it causes the wireless network to goto 100% utilization processing these requests.

Issues:
There are two issues here:
1) I tracked one down looking a cc.arduino.packages.discoverers.network.BoardReachabilityFilter's timer is too short.  5 seconds per broadcast is a bit much.  Maybe a better option would be to adjust the TTL counters in the jmdns code??
2) bugs in jmdns cause many broadcast packets (one for each service responder).  This seems like a big issue, and I logged a ticket on the jmdns git repo to see if someone can look at it.  I'll try to spend some time myself. (I tested this issue with a simple java program that just sends one start
see sample java program : [MainClass.zip](https://github.com/arduino/Arduino/files/1387692/MainClass.zip)
Just start wireshark and select this filter: "dns and udp.port eq 5353 and ip.addr eq 224.0.0.0/24" to isolate the JmDNS traffic.

Proposed Fixes:
the BoardReachabilityFilter.start should not have a fixed 5 second poll.  Although this is nice when there is only one OTA client, it gets out of hand when there are a bunch.  I suppose that if the jmdns service is fixed it would reduce this timer, but if you have a few copies if the IDE up on different computers the flooding would begin.
The jmdns service needs to be replaced, or fixed.

kurtgo, 2017-10-17:
I found a solution that will work:
1) Upgrade JmDNS to version 3.5.3.  This fixes the massive brodcast packet issues.
2) Change TTL to 10 seconds inside the new JmDNS by:
        System.getProperties().setProperty("net.dns.ttl", "10");   
3) remove the BoardReachabilityFilter timer, as it is not needed.

This solution will cause the re-broadcast as the TTL on the entries falls to zero.  It also adjusts the rebroadcast internal to the JmDNS code, so deleting and re adding the service request is not required.

facchinm, 2017-10-17:
This is a great news, thanks for sharing. Would you mind opening a PR with this fix so we can merge it directly giving you the right attribution? Thanks!

facchinm, 2017-10-19:
@kurtgo https://github.com/arduino/Arduino/compare/ide-1.9.x-beta...facchinm:jmdns_3.5.3?expand=1 implements your suggestions; it seems to work well and it will probably be included in beta1.9 for some staging tests. Do you believe it's ok?

kurtgo, 2017-10-22:
Thanks facchinm,
Yes.  Looks good to me.  Ultimately, I wish that the jmdns service had some tuning options so the TTL could be set without the dns property, but this will definitely help with packet flooding.  I'll pull this version and give it a try.

- kurt