arduino/Arduino #6833: Sketch reseting after open board manager or library manager

engeeknyer, 2017-10-17:
Just start checking the Arduino 1.9.0-beta IDE. (linux 64 bit version).

I realized that i wrote an sketch and then go to board manager, closed the manager and the sketch is gone! It reset to the initial empty setup and loop functions. The same result if I open the library manager instead of opening the the board manager.

facchinm, 2017-10-17:
Hi @engeeknyer ,
this bug was introduced yesterday (sorry about that), the next build (17) will target this problem.
It will be ready in about 10 minutes. If you could check that the bug is gone and eventually close the issue it would be great! Thanks for testing!

per1234, 2018-01-16:
@engeeknyer, did you ever get a chance to try out the latest beta build? If so, was the bug resolved?

per1234, 2018-09-26:
Closing due to lack of response.

@engeeknyer if you test the latest beta build and find the problem still occurs I'll be happy to re-open this issue.

closed 2018-09-26 by per1234