arduino/Arduino #6834: KRACK vulnerability?

sotos183, 2017-10-17:
Is the WiFi101 library of Arduino vulnerable to the KRACK vulnerability? If yes, do we have an ETA of the patch?

Thank you

cmaglie, 2017-10-17:
Microchip has released a patch today, an upgrade is coming...

http://www.microchip.com/design-centers/wireless-connectivity/embedded-wi-fi/wpa2-protocol-vulnerability

sandeepmistry, 2017-10-17:
The link above doesn't actually contain an updated firmware binary, I've contacted Microchip support for clarification on where the new firmware binary can be downloaded from.

per1234, 2017-10-17:
Shouldn't this discussion be moved to https://github.com/arduino-libraries/WiFi101/issues/194?

kbumsik, 2017-10-18:
Hi, I came from the WiFi101 library issue: https://github.com/arduino-libraries/WiFi101/issues/194.
@sandeepmistry you probably missed it. The link above does actually provides the link to download the binary `m2m_aio_3a0.bin` and documents on how to update it as well. Click "WINC15x0 Firmware Update and WPA2 Fix" in the above link page.

facchinm, 2017-10-18:
@kbumsik the zip doesn't actually provide a link to the binary, but only three documents explaining how to update. I just checked and updated every bit of Atmel Studio but the bundled firmware is still 19.5.2. Do you have any direct link by any chance?

kbumsik, 2017-10-18:
@facchinm That is weird. The zip contains two binary files and three documents.
[image]
The `FirmwareUpdate to 19.5.4 using fw_update_example v1.1.pdf` file explains you can create the old firmware update project. After you create the project, there will be the old binary file `src/firmware/firmware/m2m_aio_3a0.bin` with the binary file with the same name in the zip.

facchinm, 2017-10-18:
This is SUPER weird; the zip I downloaded only contains the pdfs (the md5 is `17a65ed946a06ee0aec9f9b1208e7239  WINC1500 Firmware Update and WPA2 Fix.zip` ).
Could you attach the bin here? Thanks!

kbumsik, 2017-10-18:
The md5 is different; mine is `394A8045E06A189BA806D710AF2493BC`. You may download it again. Maybe Microchip changed it silently? 
Anyway, I upload my zip file here.
[WINC1500 Firmware Update and WPA2 Fix.zip](https://github.com/arduino/Arduino/files/1395326/WINC1500.Firmware.Update.and.WPA2.Fix.zip)

sandeepmistry, 2017-10-19:
A PR build of the IDE with a new WiFi101 firmware updater can be found here for those wanting to try it out and provide their feedback: https://github.com/arduino/Arduino/pull/6840#issuecomment-337951622

cmaglie, 2017-10-23:
The update is now available on the hourly build, and it will be released with IDE 1.8.6.

closed 2017-10-23 by cmaglie