arduino/Arduino #6836: Arduino 1.9 on macOS 10.13 — Code complete is not available. Try increasing ulimit.

rei-vilo, 2017-10-17:
On macOS 10.13, pressing Ctrl-Space for code complete (and waiting a while) throws an error message.

`Code complete is not available. Try increasing ulimit.`

Full log:

> /private/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/AppTranslocation/1908B6D7-0CE2-4CFC-A7DD-8429EACCC25F/d/Arduino.app/Contents/Java/arduino-builder -code-complete-at /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_modified_sketch_621385/sketch_oct17a.ino:3:14 -logger=machine -hardware /private/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/AppTranslocation/1908B6D7-0CE2-4CFC-A7DD-8429EACCC25F/d/Arduino.app/Contents/Java/hardware -hardware /Users/ReiVilo/Library/Arduino15/packages -tools /private/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/AppTranslocation/1908B6D7-0CE2-4CFC-A7DD-8429EACCC25F/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/AppTranslocation/1908B6D7-0CE2-4CFC-A7DD-8429EACCC25F/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ReiVilo/Library/Arduino15/packages -built-in-libraries /private/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/AppTranslocation/1908B6D7-0CE2-4CFC-A7DD-8429EACCC25F/d/Arduino.app/Contents/Java/libraries -libraries /Users/ReiVilo/Documents/Projets/Arduino/libraries -fqbn=arduino:samd:arduino_zero_edbg -ide-version=10900 -build-path /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_build_481988 -warnings=none -build-cache /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_cache_219975 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/bossac/1.7.0 -prefs=runtime.tools.CMSIS.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.openocd.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static -prefs=runtime.tools.CMSIS-Atmel.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -prefs=runtime.tools.arduinoOTA.path=/Users/ReiVilo/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.0 /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_modified_sketch_621385/sketch_oct17a.ino
> autocomplete failure output:
> 
> com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input
>  at [Source: ; line: 1, column: 1]
> 	at com.fasterxml.jackson.databind.JsonMappingException.from(JsonMappingException.java:148)
> 	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3781)
> 	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
> 	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2726)
> 	at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:99)
> 	at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
> 	at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
> 	at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
> 	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1663)
> 	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
> 	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
> 	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
> 	at java.awt.Component.processEvent(Component.java:6310)
> 	at java.awt.Container.processEvent(Container.java:2236)
> 	at java.awt.Component.dispatchEventImpl(Component.java:4889)
> 	at java.awt.Container.dispatchEventImpl(Container.java:2294)
> 	at java.awt.Component.dispatchEvent(Component.java:4711)
> 	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
> 	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:806)
> 	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1074)
> 	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:945)
> 	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:771)
> 	at java.awt.Component.dispatchEventImpl(Component.java:4760)
> 	at java.awt.Container.dispatchEventImpl(Container.java:2294)
> 	at java.awt.Window.dispatchEventImpl(Window.java:2746)
> 	at java.awt.Component.dispatchEvent(Component.java:4711)
> 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
> 	at java.awt.EventQueue.access$500(EventQueue.java:97)
> 	at java.awt.EventQueue$3.run(EventQueue.java:709)
> 	at java.awt.EventQueue$3.run(EventQueue.java:703)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
> 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
> 	at java.awt.EventQueue$4.run(EventQueue.java:731)
> 	at java.awt.EventQueue$4.run(EventQueue.java:729)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
> 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
> 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
> 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
> 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
> 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
> 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
> 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
>

rei-vilo, 2017-10-17:
`ulimit`gives

```
$ ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
file size               (blocks, -f) unlimited
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 256
pipe size            (512 bytes, -p) 1
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 1418
virtual memory          (kbytes, -v) unlimited
```

facchinm, 2017-10-18:
Hi @rei-vilo ,
we added the popup (not very informative, indeed) to double check if the error in the preprocessor is due to a bug or the limit of watchable files has been reached.
We use a watcher to make sure that there is no change in the backing storage which could invalidate the cache; certain programs (like Sublime) use a lot of watchers so we can't use this feature and forcefully disable the autocomplete feature.
We are thinking about a fallback without using the cache but it would be painfully slow, so in the meantime you could try increasing the limits using some hints from https://superuser.com/questions/433746/is-there-a-fix-for-the-too-many-open-files-in-system-error-on-os-x-10-7-1

rei-vilo, 2017-10-18:
Hi!

Thank you for your prompt answer, explanation and tips. 

Unfortunately, I wasn't able to circumvent the issue. Even with very high numbers like `ulimit -n 20480` and starting Arduino on the shell, Arduino crashes. `sudo lsof -n | cut -f1 -d' ' | uniq -c | sort | tail` gives `10243 arduino-b`, *i.e.* 10243 files opened by Arduino. Could the IDE modify this parameter as per https://apple.stackexchange.com/a/116754?

facchinm, 2017-10-18:
I don't think we can change the limit programmatically from the app since most of the times it launches without admin privileges. 10243 files are a lot, do you have any particular core installed?

rei-vilo, 2017-10-18:
I performed a fresh install with AVR and SAMD boards packages and a new empty sketchbook folder. Then code complete succeeded. 

However, when keeping the same limited AVR and SAMD boards packages but adding my initial sketchbook folder (*ca.* 120 libraries), code complete fails.

So it looks like code complete works only with a very limited number of installed boards packages and libraries.

facchinm, 2017-10-18:
This is totally unexpected; I have a huge sketchbook too (~150) but the number of watched dirs is much smaller. Probably there are only a bunch of libraries with massive nesting that make that number exploding. Could you check if any of your libraries is structured like this? Thanks a lot for your help debugging this issue!

rei-vilo, 2017-10-18:
One library was massive (2083 files in directory ./chipKIT-DisplayCore). Removing it allows indexing to succeed. Would the code complete indexing process rather ignore too large libraries instead of stopping the whole procedure?

facchinm, 2017-10-20:
We modified the builder to explicitly drop the cache, so the watcher is not needed anymore. If you could test the latest version it would be great!

rei-vilo, 2017-10-20:
The latest version worked great: no more messages and no more limits on boards and libraries.

Is there a way to have code complete without pressing Ctrl-Space every time?