arduino/Arduino #6842: Error when zooming out in Arduino IDE

baboundu81, 2017-10-20:
Hi everybody,

I juste wanted to report u bug I found by accident on the Arduino IDE.
When zooming out using CTRL + mouse wheel, I accidentally went too far and the font went  verry small.
I wasn't able to zoom in back to a usable font size. (I saw that there was a java error in the console window but I forgot to save it, sorry).

So I decided to quit the IDE and relaunch it but the splash screen appeared and then nothing.
I uninstalled Arduino IDE and reinstalled it but the issue was still there.

The solution was to edit C:\Users\MyUser\AppData\Local\Arduino15\preferences.txt
The font size value was -154 in the line:
`editor.font=Monospaced,plain,-154`
So I changed it to 12
`editor.font=Monospaced,plain,12`
And it worked nicely.

I found a solution but the bug still exists

Hoping to help you.

Bye !

per1234, 2017-10-20:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6689

This issue has already been reported many times and is also already solved in the Arduino IDE 1.8.5 release.

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

Please always test using the [current beta build](http://www.arduino.cc/en/Main/Software#beta) of the IDE before reporting an issue to be sure that it was not already fixed.

closed 2017-10-20 by per1234

Andy2No, 2019-06-05:
> 
> 
> Closing as duplicate of #6689
> 
> This issue has already been reported many times and is also already solved in the Arduino IDE 1.8.5 release.
> 

I've read that this has been resolved, but that has not been my experience.  In 1.8.5 I had the problem that Ctrl+Scroll (with the mouse wheel) zooms out, but not in.  I upgraded to 1.8.9 today and still have the same problem.

Since this issue has been marked as a duplicate, more than once, I'm not clear where to post about it.  Presumably, if I started a new issue thread, that would be marked as a duplicate too?

Either the problem was not fixed, or it has been reverted.

In 1.8.9  I see the keyboard shortcuts work and the one to zoom in appears in the menu correctly.  In 1.8.5 it was shown as "unknown key code".

Perhaps just fixing the keyboard shortcuts was seen as a resolution of the problem?  I habitually zoom windows with the mouse, and often don't have a keyboard in front of me.  Perhaps this was overlooked because the person fixing it habitually does the opposite?

Since I am able to zoom out with the mouse, I assume the intention was to allow me to zoom in or out.  Attempting to zoom in makes the window flicker rapidly between two similar font sizes, but doesn't have the desired effect.

So, I am still able to zoom out to a level where the text is completely unreadable, using the mouse, but not able to zoom back in again.

per1234, 2019-06-05:
>I've read that this has been resolved, but that has not been my experience. In 1.8.5 I had the problem that Ctrl+Scroll (with the mouse wheel) zooms out, but not in. I upgraded to 1.8.9 today and still have the same problem.

You're experiencing a completely different issue. This issue is about a bug where you could zoom out all the way to font size zero, which would then break the IDE. That issue has indeed been fixed.

I know I've seen a report of your issue before, but I can't find it in the issue tracker. Maybe it was on the Arduino Forum? Since we don't have an existing issue report for your issue, you are welcome to open one if you install the [current beta build of the Arduino IDE](http://www.arduino.cc/en/Main/Software#beta) and find that the problem still occurs. Make sure to provide all relevant information in the issue report, including the build number of the Arduino IDE beta you're using and your OS.

I've checked Arduino IDE 1.8.9 on Windows 10 and Ubuntu and I can't reproduce your issue.

brianmwebb, 2020-06-21:
> Hi everybody,
> 
> I juste wanted to report u bug I found by accident on the Arduino IDE.
> When zooming out using CTRL + mouse wheel, I accidentally went too far and the font went verry small.
> I wasn't able to zoom in back to a usable font size. (I saw that there was a java error in the console window but I forgot to save it, sorry).
> 
> So I decided to quit the IDE and relaunch it but the splash screen appeared and then nothing.
> I uninstalled Arduino IDE and reinstalled it but the issue was still there.
> 
> The solution was to edit C:\Users\MyUser\AppData\Local\Arduino15\preferences.txt
> The font size value was -154 in the line:
> `editor.font=Monospaced,plain,-154`
> So I changed it to 12
> `editor.font=Monospaced,plain,12`
> And it worked nicely.
> 
> I found a solution but the bug still exists
> 
> Hoping to help you.
> 
> Bye !
> 
> Edit : here is the java error in the console window
> `Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: Font size must be > 0 at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986) at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510) at processing.app.EditorTab.applyPreferences(EditorTab.java:322) at processing.app.Editor.applyPreferences(Editor.java:456) at java.lang.Iterable.forEach(Iterable.java:75) at processing.app.Base.handleFontSizeChange(Base.java:1861) at processing.app.EditorTab.mouseWheelMoved(EditorTab.java:190) at java.awt.Component.processMouseWheelEvent(Component.java:6621) at java.awt.Component.processEvent(Component.java:6305) at java.awt.Container.processEvent(Container.java:2236) at java.awt.Component.dispatchEventImpl(Component.java:4889) at java.awt.Container.dispatchEventImpl(Container.java:2294) at java.awt.Component.dispatchEvent(Component.java:4711) at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888) at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4554) at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466) at java.awt.Container.dispatchEventImpl(Container.java:2280) at java.awt.Window.dispatchEventImpl(Window.java:2746) at java.awt.Component.dispatchEvent(Component.java:4711) at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758) at java.awt.EventQueue.access$500(EventQueue.java:97) at java.awt.EventQueue$3.run(EventQueue.java:709) at java.awt.EventQueue$3.run(EventQueue.java:703) at java.security.AccessController.doPrivileged(Native Method) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90) at java.awt.EventQueue$4.run(EventQueue.java:731) at java.awt.EventQueue$4.run(EventQueue.java:729) at java.security.AccessController.doPrivileged(Native Method) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80) at java.awt.EventQueue.dispatchEvent(EventQueue.java:728) at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201) at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116) at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105) at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93) at java.awt.EventDispatchThread.run(EventDispatchThread.java:82) Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: Font size must be > 0 at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986) at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510) at processing.app.EditorTab.applyPreferences(EditorTab.java:322) at processing.app.Editor.applyPreferences(Editor.java:456) at java.lang.Iterable.forEach(Iterable.java:75) at processing.app.Base.handleFontSizeChange(Base.java:1861) at processing.app.EditorTab.mouseWheelMoved(EditorTab.java:190) at java.awt.Component.processMouseWheelEvent(Component.java:6621) at java.awt.Component.processEvent(Component.java:6305) at java.awt.Container.processEvent(Container.java:2236) at java.awt.Component.dispatchEventImpl(Component.java:4889) at java.awt.Container.dispatchEventImpl(Container.java:2294) at java.awt.Component.dispatchEvent(Component.java:4711) at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888) at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4554) at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466) at java.awt.Container.dispatchEventImpl(Container.java:2280) at java.awt.Window.dispatchEventImpl(Window.java:2746) at java.awt.Component.dispatchEvent(Component.java:4711) at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758) at java.awt.EventQueue.access$500(EventQueue.java:97) at java.awt.EventQueue$3.run(EventQueue.java:709) at java.awt.EventQueue$3.run(EventQueue.java:703) at java.security.AccessController.doPrivileged(Native Method) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90) at java.awt.EventQueue$4.run(EventQueue.java:731) at java.awt.EventQueue$4.run(EventQueue.java:729) at java.security.AccessController.doPrivileged(Native Method) at java.security.ProtectionDomain$JavaSecurityAccessImpl.doInterse [... cut ...] ue.dispatchEvent(EventQueue.java:728) at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201) at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116) at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105) at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93) at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`