arduino/Arduino #6848: Please add these libraries to the manager.

Trusty77, 2017-10-23:
https://git.framasoft.org/locoduino.org/Accessories
https://git.framasoft.org/locoduino.org/Commanders
https://git.framasoft.org/locoduino.org/DIO2
https://git.framasoft.org/locoduino.org/LcdUi
https://git.framasoft.org/locoduino.org/EEPROMExtent
https://git.framasoft.org/locoduino.org/MemoryUsage
https://git.framasoft.org/locoduino.org/CommandInterpreter
https://git.framasoft.org/locoduino.org/SlowMotionServo

cmaglie, 2017-10-23:
The library manager only supports repositories from github, bitbucket or gitlab: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager
If you want to add the libraries you must move or create a copy in one of them.

Are these libraries part of a framework? if, yes, you may want to consider prefixing the name with something like `locoduino_`, so `locoduino_Accessories`, `locoduino_Commanders`, etc. so they don't get mixed sparse in the 1000+ libraries in the library manager.

Trusty77, 2017-11-23:
Could you add these libraries to the manager ?

https://github.com/Locoduino/Accessories
https://github.com/Locoduino/Commanders
https://github.com/Locoduino/DIO2
https://github.com/Locoduino/LcdUi
https://github.com/Locoduino/EEPROMExtent
https://github.com/Locoduino/MemoryUsage
https://github.com/Locoduino/DCCpp
https://github.com/Locoduino/DcDccNanoController

Thank you.

per1234, 2017-11-24:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Trusty77, 2017-11-24:
Sorry for the mistakes. I think i have done all the necessary things...
Could you add now these libraries to the manager ?

https://github.com/Locoduino/Accessories
https://github.com/Locoduino/Commanders
https://github.com/Locoduino/DIO2
https://github.com/Locoduino/LcdUi
https://github.com/Locoduino/EEPROMExtent
https://github.com/Locoduino/MemoryUsage
https://github.com/Locoduino/DCCpp
https://github.com/Locoduino/DcDccNanoController

Thank you.

per1234, 2017-11-24:
There are two where the `version` value in library.properties doesn't match the release version:
- https://github.com/Locoduino/Commanders/blob/v1.51.1/library.properties#L2
- https://github.com/Locoduino/DCCpp/blob/v0.5.1-beta1/library.properties#L2

Trusty77, 2017-11-24:
All should be right now...

per1234, 2017-11-24:
DCCpp is fixed but the update of the `version` value in the Commanders library was done after the v1.51.1 release so the `version` value in the v1.51.1 release is still 1.51.0:
https://github.com/Locoduino/Commanders/blob/v1.51.1/library.properties#L2

Trusty77, 2017-11-24:
Should be better now. An unwanted manually created tag... I am almost new to git, and really new to github !
Thanks for your support.

cmaglie, 2017-12-04:
Done!

closed 2017-12-04 by cmaglie