arduino/Arduino #6850: Error in pins_arduino.h - Mega2560 Timer1C PWM output

sgm23, 2017-10-23:
There is an error in this file as found in the variants\mega subfolder of the install directory that prevents the use of the TIMER1C (OC1C) output on the Mega2560.
Specifically it is missing the line TIMER1C , // PB 7 ** 13 ** PWM13, I have spent the last week trying to get this third PWM output from timer C in frustration as i could not see there my code was wrong. Clearly is was not, please correct this for the next distribution.

matthijskooijman, 2017-10-23:
In the [pins_arduino.h]() file, there is this line:

	TIMER0A , // PB 7 ** 13 ** PWM13

According to the datasheet, the same pin is used as both OC0A and OC1C, so it seems PWM is available on this pin by using analogWrite already, it just uses timer0 instead of timer1 (but you shouldn't really care about that when using analogWrite).

Can you clarify a bit more why you think this would need to be changed?

sgm23, 2017-10-24:
This needs to be changed for several reasons…
The datasheet for the ATmega2560 says you can get a PWM output from TIMER1C (OC1C) on that pin therefore you should be able to do so and as far as I can tell it is impossible to do that without this change.

That alone should be reason for the change, people will design their PCB and pin assignment assuming they can do that and others should not have to subsequently waste the hours I have trying to find the bug in bug free code or redesign their project just because the device is crippled by some easily corrected omission in a file.

However if that is not enough reason for you…
Not all timers are equal.
Timer 0 is 8 Bit, Timer 1 is 16 Bit, consequently they require different setups and different values sent to them to obtain the same duty cycle. That is a lot of unnecessary effort to force people to go to for the sake of such a quick and simple change to a file, thus using Timer0, particularly when trying to drive a 3 phase motor is somewhat unhelpful because…
a)
Being driven from a separate timer to the other two phases mean the phases will not be in sync, unlike generating all three phases from a single timer.
This will result in a small amount of undesired braking of the motor resulting in loss of energy/speed, back emf spikes and a small amount of jitter in the motors rotation.
b)
Other people like me may want/need to use the micros and/or millis commands from the Timer.h include. Which clearly indicates it uses Timer0 and not to use Timer0 for anything else as it can mess up returned values. 

Thus using Timer0 is not always a suitable alternative because it is not the same thing!

analogWrite:
I am glad you mentioned analogWrite in your reply as that also requires some modification in my opinion…
To write that command in a way that 255 is a special case and always result in 100% duty cycle in a PWM signal was rather short sighted even if 8 bit timers were only available at the time as is assumes TOP will always be 255 which is not necessarily the case as it may be set to a value lower value to facilitate a PWM frequency that is different from the default.
For 16 bit timers where TOP may be, as in my case set to 400 to produce a 20KHz PWM signal the special case for 255 creates a problem that then has to be worked around as otherwise it results in 100% duty cycle at a time when that is not what you want.
Consequently either the special case for 100% duty cycle should be removed and all non zero values treated the same or the special case for 100% should take its value from that set by a new "**analogWriteTop**" command so that it works correctly for values other than 255. 

Regards,
Stephen

matthijskooijman, 2017-10-24:
> The datasheet for the ATmega2560 says you can get a PWM output from TIMER1C (OC1C) on that pin therefore you should be able to do so and as far as I can tell it is impossible to do that without this change.

There is a lot of things that the AVR hardware can do, that you cannot do through the Arduino API. However, if you write to the AVR hardware registers directly, you should be able to use pretty much everything, *including* the OC1C output, without any changes to the pin mapping.

As for the 8- vs 16-bit timers, `analogWrite()` only offers 8-bit resolution (IIRC it configures the 16-bit timers to in 8-bit mode), so hardcoding 255 is sensible. If you need more, you'll have to talk to the hardware registers directly anyway.

As for PWM outputs being in phase, this is not something the analogWrite API guarantees in any way, it's just a side effect of the internal implementation. Even then, it makes no sense to change the pin to be in phase with the other timer1 pins, since it would then no longer be in phase with the timer0 pins as it is now.

> Timer 0 is 8 Bit, Timer 1 is 16 Bit, consequently they require different setups and different values sent to them to obtain the same duty cycle.

This is already handled by `analogWrite()` (together with `init()`), so no additional effort is required from the user.

Basically, you have two choices:
 1. You use `analogWrite` to generate PWM output. In this case, you get 8-bit, arbitrary phased PWM output with an easy interface.
 2. You use the hardware registers directly (to set up the timer, set up duty cycle and to assign PWM outputs to pins using the COMxxx bits) and *not* use `analogWrite()` at all. In this case, you can use the full 16-bit on some timers, use arbitrary top values, get phase-aligned outputs, etc. Since you're not using `analogWrite()`, the values in `pins_arduino.h` do not matter at all.

It seems to me that you are trying to mix both choices, which simply is not a usecase that can be easily supported.

As for changing the pin assignment for pin 13 from timer0a to timer1c, I do not see any specific advantages for this, within the supported usecases (1. and 2. above). Either assignment would work, so even though there's no real advantage to the current assignment either, I'd rather not change it since that might break people's code in some corner cases.

sgm23, 2017-10-24:
> As for PWM outputs being in phase, this is not something the analogWrite API guarantees in any way, it's just a side effect of the internal implementation. Even then, it makes no sense to change the [...]

I think we must have a communication problem somewhere as that statement make no sense to me.
I want the output on that pin in phase with the other timer 1 pins, that is the whole point (I am not using any Timer0 outputs on a pin). They should be in phase because they are using the same timer as a source. Perhaps it will make more sense if you think of it this not quite analogous way but…
You buy a car, how uncontrollable would that car be if the front wheel are in phase, turning in the same direction at the same speed, the rear wheels are turning at a different speed – not a desirable situation! 

> Basically, you have two choices:
> 1.	You use analogWrite to generate PWM output. In this case, you get 8-bit, arbitrary phased PWM output with an easy interface.
> 2.	You use the hardware registers [...]

As the code below (in conjunction with the requested file change) appears to be to be the only way I can find that generates the three mode 10 (Phase Correct PWM) outputs I want from Timer1 even though you appear to be saying that it should not because I am mixing two incompatible methods, can you please explain to me how do I achieve that using the non analogWrite() – hardware registers directly method? 
As when I comment out the analogWrite() commands in my code and write to the OCRnx registers, which to the best of my knowledge is what will determine when PWM signal changes from High to Low upon a value match with the timer count I get no PWM output at all!

```
//Load required Includes - Complete 31/03/2017
#include <avr/wdt.h>    //Include for Watchdog Timer - sgm June 2017

//*** Assign Alias's to Arduino board I/O pin numbers *** - complete 06/06/2017
//MOTOR CONTROL
#define PWM_SPARE 11  //Spare PWM output (OC1A)
#define DCM_PWM 12    //DC Load Machine PWM/Inhibit (OC1B) - rename DC PWM.
#define PWM 13        //PWM source for single BLDCM phase PWM signal / Allways high gate for individual BLDCM phase PWM signals (OC1C) 

void setup() {

TCCR0A = TCCR0A & 0x3F;  //Set COM0A1 & COM0A0 to 0, disable Timer OCOA output so that only Timer1 output is on Digital Pin 13 (PB7)
TCCR1A = 0xAA;           //Set COM1A1, COM1B1, COM1C1 & WGM11 to 1 (1010 1010) = Phase Correct PWM, TOP = ICRn, output = Timer1 A, B & C
TCCR1B = 0x11;           //Set WGM13 & CS10 to 1 for PWM with no Prescaling 0001 0001 - DO NOT CHANGE bit 7:3 of this value, only change bits 2:0 (prescalling)
TCCR1C = 0x00;           //Ensure TRRC1C is clear, TCCR1C not used, just belt & braces code
ICR1 = 400;              //Use IRCn to Set TOP for a 20KHz PWM Frequency  = 400 ($0190) 

//Set up initial state for Infineon Motor drivers & PWM Signal- Complete 07/06/2017
digitalWrite(PWM_SPARE, LOW);  //No PWM - Output in Tristate condition
digitalWrite(DCM_PWM, LOW);    //No DC Load Machine PWM - Output in Tristate condition
digitalWrite(PWM, LOW);        //No PWM - Output in Tristate condition

//*** Set up I/O status of Infineon Motor driver control & PWM output pins ***
//Set DC Load Machine PWM control pins to Outputs (Digital Pins) - Complete 06/06/2017
pinMode(PWM_SPARE, OUTPUT);  //Set pin to Output
pinMode(DCM_PWM, OUTPUT);    //Set pin to Output
pinMode(PWM, OUTPUT);        //Set pin to Output

//Set Modulation depth (Testing only)
analogWrite(PWM_SPARE, 300);  //75% Pulse width on Timer1A (OC1A)
analogWrite(DCM_PWM, 200);    //50% Pulse width on Timer1B (OC1B)
analogWrite(PWM, 100);        //25% Pulse width on Timer1C (OC1C)

//OCR1A = 300;  //Generate no PWM
//OCR1B = 200;  //Generate no PWM
//OCR1C = 100;  //Generate no PWM
WDTO_2S;  //2 Second Time out (Watchdog Timer enabled at power up)
}
// END OF SETUP

//=========================
//=== Main Program loop === does not return data (Void) - basically done needs a few tweeks
//=========================
void loop() {

//Reset Watchdog Timer (AVR/wdt.h)
wdt_reset();
//asm volatile("wdr");  //Reset Watchdog Timer using Assembler command - I hope?
}
//=== END OF Main Program Loop ===
```

matthijskooijman, 2017-10-24:
> I think we must have a communication problem somewhere as that statement make no sense to me.

My point is that `analogWrite()` is an API, that makes certain guarantees about how it works. For example, if you pass 128, into it, you'll get a 50% duty cycle PWM signal on the pin. However, it makes no guarantees about the phase of the PWM cycle. In practice, due to the fact that multiple pins share the same timer, some pins will be in phase, but this is not something that is part of the interface (it will likely be different on different architectures, for example).

>  As when I comment out the analogWrite() commands in my code and write to the OCRnx registers, which to the best of my knowledge is what will determine when PWM signal changes from High to Low upon a value match with the timer count I get no PWM output at all!

In addition to writing the OCRnx register, you must also set the relevant COMnxx pins to enable PWM output on the associated pins. The datasheet can tell you more, or you can look at [how `analogWrite()` does this](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring_analog.c#L166-L170).

matthijskooijman, 2017-10-24:
Since I do not think this is a change that is going to be made in the Arduino core, and since you should have enough info to get your code working now, I'm closing this issue. Feel free to leave additional comments if needed.

closed 2017-10-24 by matthijskooijman

PaulStoffregen, 2017-10-24:
> a new "analogWriteTop" command so that it works correctly for values other than 255.

Years ago, after *much* discussion on the public mailing list, it was decided the function analogWriteFrequency(pin, frequency) would be used when/if the core library ever supported different top & prescale values.

Arduino never actually implemented this in any of their core libraries (avr, sam, samd).  I'm not saying Arduino should do this.  I'm only posting a remind that this API was previously discussed and the decision was made, if it were ever to be done, to use analogWriteFrequency(pin, frequency).

sgm23, 2017-10-25:
> In addition to writing the OCRnx register, you must also set the relevant COMnxx pins to enable PWM output on the associated pins. The datasheet can tell you more, or you can look at how analogWrite() does this.

Not helpful as I have already done that!
`TCCR1A = 0xAA;           //Set COM1A1, COM1B1, COM1C1 & WGM11 to 1 (1010 1010) = Phase Correct PWM, TOP = ICRn, output = Timer1 A, B & C`

I will be glad to get back to PICs where people offer actual solutions and file errors/bugs are corrected without having to justify why they should be instead of this unhelpful amateur hour nonsense.

photodude, 2019-04-24:
I just stumbled into this issue, (sorry to zombify it) I have a couple of comments that I think are relevant to the justification that Arduino Core does not offer Timer 1 access for pin 13. although PB7 (aka pin 13 on the mega) has OC0A/OC1C/PCINT7 available, there are a lot of gotchas involved in trying to set up OC1C for use as a timer 1 PWM signal.
- OCR1C is disabled by default on the AVR chip (see section 17 of the AVR datasheet)
- If both OC0A and OC1C are active you don't get expected results as it ends up in Output Compare Modulator mode (see section 19 of the AVR datasheet)
- if you disable OC0A you break all of the functions that rely on OC0A ( I believe it's the internal time functions like millis(), and delay() and all functions which rely on those timing functions)

on the flip side there are some disadvantages to pin 13 being stuck on Timer 0.
- Pin 13 is stuck at the default 400hz PWM unless you modify Timer 0 (
  - 99.999% of people should never modify timer 0 as you mess up time functions like millis(), and delay() and all functions which rely on those timing functions
- access to working with Pin 13 on timer 1 would be nice as it would give control over the PWM frequency. But note the very significant issues with trying to use Pin 13 on timer 0

IMHO, it is appropriate for Arduino platform to skip the pin 13 access to the timer 1 settings on the mega by default. For very advanced users who need such access they can figure out how to deal with the registers, limitations and consequences of making such a change. Other users might also choose to implement an I^2C extension to give them more PWM pins. Furthermore, the most advanced users are going to be working directly with the chip and not with the arduino platform as such they have no need for the arduino API and are not limited by the design choices here.

Thanks for discussing the topic, it's clear that this was a design choice for the majority of users and that
very advanced users can figure out the registers to implement timer 1 on pin 13 if they so choose.