arduino/Arduino #6854: Auto Format (Ctrl+T) formats strings

JJames47, 2017-10-25:
I have been using v1.82 and have just installed v1.85 and during a program modification I used Auto Formatting.  When my program refused to work correctly I finally trace this fault to strings which had their content subject to the Auto Formatting rules.  I my case all strings containing "/" had been altered to " / " (literals not included) i.e a space had been added either side of the slash symbol.

per1234, 2017-10-25:
Please provide a minimal sketch that we can use to reproduce the issue.

JJames47, 2017-10-25:
I may have been a bit premature when I opened this.

Immediately after installation, v1,85 had run continuously for 2 days monitoring program output.   After an edit the "problem" appeared which after diagnosing I was able to repeat it twice more.  Last night I closed the IDE and shut down the PC.  This morning I filed the issue and after receiving your response I reloaded the IDE and the "offending" program but the problem with the Auto Format does not now occur.

cmaglie, 2017-10-25:
@JJames47 
I'm closing the issue for now, please comment further (even if the issue is closed) when you have a sketch that helps to reproduce the problem.

closed 2017-10-25 by cmaglie