arduino/Arduino #6859: PUSB: Possible to initialize multiple EPs under a single interface?

dogtopus, 2017-10-26:
I want to have an IN EP and an OUT EP under the same interface, but when I write 
```C++
MyDevice_::MyDevice (void): PluggableUSBModule(2, 1, epType) {
  epType[0] = EP_TYPE_INTERRUPT_IN;
  epType[1] = EP_TYPE_INTERRUPT_OUT;
}
```
for example (epType has size of 2), the pluggedInterface and pluggedEndpoint are all set to 0, and the OS fails to detect my interface (not showing up on `lsusb -vd VID:PID`).

Using `numEps = 1` and `numIfs = 1` the OS detects my interface but I know it will not work properly.

facchinm, 2017-10-26:
Hi @dogtopus ,
this is kinda strange (I'll test it as soon as I can). `PluggableUSBModule(2, 2, epType) ` definitely works (we use it on MIDIUSB library (https://github.com/arduino-libraries/MIDIUSB/blob/master/src/MIDIUSB.cpp#L190) and the cons of having one more interface should be not that terrible. Probably there is a bug somewhere that creates wrong descriptors.

dogtopus, 2017-10-26:
Well, I think I found the problem.
I was coding for a 16u2, and I didn't know that it has only 4 programmable EPs, and the CDC serial interface in the Arduino library already consumes 3 of them.

closed 2017-10-27 by facchinm