arduino/Arduino #6860: Bazel Support

jeremyje, 2017-10-26:
It'd be nice if the Arduino compile had Bazel (http://bazel.build) integration.

Something like:

```
arduino_cc_binary(
  name = "example",
  srcs = ["example.ino"],
  board = "@io_github_arduino_arduino//boards:esp8266",
  libraries = ["rs232"],
)
```
This would download the board definition, arduino compiler, and required libraries and build them to bazel-bin/example.bin

hblok, 2018-06-15:
This might be relevant: 
https://opensource.google.com/projects/platformio_rules

Kernald, 2019-03-29:
These PlatformIO rules are working fine (I opened a few issues this past month, and the developer has been reactivate to fix them, that's nice), but have a few issues inherent to the way they're built: they're using PlatformIO as a kind of middleware, which isn't the way it's meant to be used. You cannot leverage any standard C/C++ feature that Bazel have, re-using modules with other code means writing a build target using cc_library, and a second one using platformio_library (for the same source set), it also requires to have PlatformIO in your path, which makes reproducibility hit-and-miss. In addition to that, PlatformIO's dependency management is great... for simple cases. It's a mess with nested folders.

While it's great to have those rules (they aren't perfect, but they work), I would love to see some board-specific C/C++ toolchains, native definition of Arduino libraries (Wire...), and a simple rule to manage the upload.

sascha1337, 2020-01-31:
pump this up ! when bazel ?

victorromeo, 2020-06-16:
Agreed.  Love to see Bazel supported here in a more independent way. Arduino's linker isn't great when referencing libraries outside Arduino IDE paths.

simonhorlick, 2021-06-16:
I spent a bit of time adding a Bazel toolchain that works for the ESP32. It's a bit rough around the edges, but it might be useful: https://github.com/simonhorlick/bazel_esp32

sascha1337, 2021-06-16:
hah mate, been a while since askin, but better late than never right ? Thanks mate.

henriksod, 2023-07-17:
Bump! Currently looking for this.