arduino/Arduino #6867: Creators update: >Compile errors after document folder moved to \onedrive\ subfolder

phil85521, 2017-10-27:
I had a working arduino IDE.

After I updated Windows to 1709 creators update, I suddenly couldn't compile anymore. It always stopped and told "Couldn't find file...".
After searching I found out that Windows moved the documnent directory below the onedrive path.
Arduino seems to have issues with this path.
I moved the project to another directory path and the program compiled as normal again.
Arduino IDE version 1.8.5 from Windows store.
Before I updated yesterday I used 1.6.10 version from download area. Both were showing same issue.

I don't know if during update of Windows I was asked if I wanted the document folder to be moved.

per1234, 2017-10-27:
This was also reported here: https://github.com/arduino/arduino-builder/issues/90#issuecomment-338495219

matthijskooijman, 2017-11-16:
I've opened https://github.com/arduino/arduino-builder/issues/254 for this, so I'm closing this issue.

closed 2017-11-16 by matthijskooijman