arduino/Arduino #6870: sending data by serial port

sebacalka, 2017-10-29:
Hello

The problem is that the arduino (I have uno and leonardo behave the same) start the serial port after sending the data correctly, but can not properly receive it.I decided that if we send one digit, 

Arduino receives: this digt + 48.
example: we send 2, arduino recive 50; we send 0 arduino recive 48

If we send 2 or more digits we will get each of the numbers added 48.
example: we send 24, arduino recive 5052; we send 305, arduino recive 51,48,50

The transmission speed is the same for PC and 
Aruino IDE reinstalling the environment did not help.

What could be wrong?

per1234, 2017-10-29:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2017-10-29 by per1234

PaulStoffregen, 2017-10-29:
When you post on the forum, put a complete copy of your program in the message.  Use the code tags to do so.  Also give people a screenshot of the wrong data.  Nobody can help you solve this problem without complete info.

mikaelpatel, 2017-10-29:
@sebacalka 

> What could be wrong?

What you are experiences is the difference between representations of numbers. 48 is the decimal number for the ASCII character '0'.